commaai/openpilot #225: How to capture the video on neo

qmind, 2018-03-25:
In [7]: vd = open('/dev/video0', 'rw')

In [8]: cp = v4l2.v4l2_capability()

In [11]: fcntl.ioctl(vd, v4l2.VIDIOC_QUERYCAP, cp)
--------------------------------------------------------------------------
IOError                                  Traceback (most recent call last)
<ipython-input-11-890906fb3651> in <module>()
----> 1 fcntl.ioctl(vd, v4l2.VIDIOC_QUERYCAP, cp)

IOError: [Errno 25] Not a typewriter

Does anybody know how to capture the video stream on neo?

richardhughes260, 2018-03-25:
Opp  0 ppl

pd0wm, 2019-06-02:
I'm not sure what you're trying to do. Feel free to reopen with a more elaborate explanation of what you're trying to achieve and what doesn't work.

closed 2019-06-02 by pd0wm