commaai/openpilot #227: Access to the vision algorithm

hrubaiyat, 2018-03-26:
Hi, the vision algorithm used in openpilot is not open. Is it possible to make it open for PC version or at least provide a similar but simpler algorithm to mimic the actual visiond in PC? Thanks.

vanillagorillaa, 2018-03-27:
Visiond is closed source for business model reasons.

hrubaiyat, 2018-03-29:
I understand. Is it possible to know what is the frame rate of the visiond. I know the rate of main control loop is 100. But the vision should be slower than this. Thanks

Glidepathuav, 2018-03-30:
That being said we cannot use anything other than Motorola One plus to Run open Pilot?

vanillagorillaa, 2018-03-30:
OpenPilot only runs on the OnePlus 3/3t. It is made by OnePlus

Glidepathuav, 2018-03-31:
Vanilla guerilla

Understood! Thank you for replying.

The motorola one plus family has evolved a bit since One plus 3. Can we use
the one plus 5 or moto Z?

Thanks in advance

Eric H

energee, 2018-03-31:
@geohot please close