commaai/openpilot #229: Openpilot "Resume" like Stock ACC

Jamezz, 2018-03-30:
(Planned on implementing this myself, but wanted to get feedback if there was a specific reason why this wasn't done)

In some of the stock (2017 Volt) ACC systems, pressing the Resume button engages the ACC back to the originally set maximum cruise speed instead of overwriting it with the current vehicle speed.

Would be great to have OP do the same thing so drivers didn't have to readjust the speed every time OP needs to be re-engaged.

geohot, 2018-03-30:
That doesn't work? What car?

Jamezz, 2018-03-31:
This is an issue with the 2017 Volt port. Code inspection led me to believe that this was intended in OP (accelCruise/decelCruise event resets speed if OP state transitions from Disengaged->Engaged), but your reply seems to imply otherwise. Maybe this might just be a bug downstream in the volt port?

Jamezz, 2018-07-22:
Fixed in 0.5! Resume to previously set speed is so nice...

closed 2018-07-22 by Jamezz