commaai/openpilot #233: Dose EON Kit support Janpanse version's Toyota 2016 RAV4 with TSS-C ?

cruise0, 2018-04-10:
Hello , Everyone
 I have setup EON Kit on my Toyota 2016 RAV4 with TSS-C (which is a Japanese ).
 The software version of my EON Dashcam is v0.4.0.2 

1.  My EON Dashcam can't enter the lane track mode unless plug usb power cable to EON Dashcam.

2. However if I want to  pair my RAV4 with EON, the panda also need to connect to EON with the same usb port.

Here is my problem , if I chose to pair my car (EON connect to panda) then it will never enter lane detect mode on screen, unless unplug the usb-cable to panda and plug 5V usb power cable to EON. So I cannot perform the initial calibration now.

Is there anything wrong on my setup ? or Does EON kit support Toyota 2016 RAV4 with TSS-C ?

vanillagorillaa, 2018-04-10:
Openpilot does not work with TSS-C

vanillagorillaa, 2018-04-10:
you can however do ChffrPlus if you wish

cruise0, 2018-04-10:
Thanks , vanillagorillaa.
You provide me a very useful information.

That me update my current status. It looks like I need the [comma power](https://shop.comma.ai/products/power) to charge my giraffe and EON Dashcam from RAV4's ODBII.

After get extra power form ODBII to giraffe and Dashcam. I can perform the initial calibration to 100%.

However, after 100% calibration, the EON Dashcam still display "Disengaged".
 
Is it caused by using 2016 RAV4 TSS-C version ? 
So, If I have a TSS-P RAV4,  then EON can display "Engaged" and perform lane keeping assist ?

nxarmada, 2018-04-13:
you need TSS-P w/ 'Full Speed Range Dynamic Radar Cruise Control'

closed 2018-05-15 by cruise0