commaai/openpilot #238: calibration tool?

myf, 2018-04-30:
was just trying to test out openpilot 0.4.2 with grey panda, after calibration it seems that our car got really close to the right lane marker and sometimes we would ride on top of the lane. People suggested that the position of the NEO is very important. We simply eye-balled it instead of calibration. Is there any calibration tool that comes with and if not, would there be interests in making one since it's kinda crucial to the whole system?

Glidepathuav, 2018-05-01:
You will need to fabricate a fixture to help aid in perfect install in
order to make it all jive.

vanillagorillaa, 2018-05-01:
Just going to drop this here .... 

https://community.comma.ai/wiki/index.php/Gridded_UI

myf, 2018-05-01:
ahh that is very relevant, thank you

closed 2018-08-16 by rbiasini