commaai/openpilot #240: Logo Design

nunojesus, 2018-05-03:
Hi,
I have design this logo with the letter O and P, if you like you can keap it for this project.
If you want it I will give you all the formats of the design free. If you want to make changes please specify.

Greetings

[image]

joshuairl, 2018-05-03:
I love it.  Its simple but very brilliant.  I support it.

nunojesus, 2018-05-03:
Thanks, what color do you prefer as the main logo, I can send you also the other colours too.
Can I make the png files to send to you?
Greetings!

nunojesus, 2018-05-03:
See if you like the font type

[image]

mtourangeau, 2018-05-03:
What about this? I did not complet the O because I'm not good at using MS Paint :)
[image]

joshuairl, 2018-05-03:
Not digging the type.  @mtourangeau That's not too shabby either.

nunojesus, 2018-05-03:
in my opinion the logo that I presented has a stronger presence, but you decide what is best for the project.

nunojesus, 2018-05-10:
Since I dont have feedback from you I will close this issue. Good luck for your project.

closed 2018-05-10 by nunojesus