commaai/openpilot #241: 2018 Sonata w/LKAS + Collision avoidance

Brandon-Myers, 2018-05-04:
I have a 2018 Sonata with LKAS and the front radar for adaptive cruise control and collision avoidance.

What would it take for me to start a pilot program to have this car (And other Hyundai's with the same equipment) supported?

Brandon-Myers, 2018-05-04:
FYI: Read the medium article about the Toyota.

vanillagorillaa, 2018-05-04:
Hyundai is not officially supported yet. You will need to do a port yourself or with others. I would join the slack and ask in the Hyundai channel. There are a few users working on a port and making decent progress.

veritas2884, 2018-07-13:
@Brandon-Myers I cannot find that slack group. Has any progress been made? I have a 2016 Sonata with Lane Departure Warning and Full Stop Adaptive Cruise Control and want to get Openpilot running on it.

energee, 2018-07-13:
[slack.comma.ai](slack.comma.ai)

veritas2884, 2018-07-15:
@energee thanks. I don't see a Hyundai subchannel though.

emmertex, 2019-01-15:
@vanillagorillaa This should be closed.
Hyundai/Kia are well supported by community, and supported officially.

closed 2019-01-15 by vanillagorillaa