commaai/openpilot #242: How to desactivate Dashcam recording? to choose if I want record or not

DaveAL, 2018-05-07:
When I go to long trip(+500 miles) outside Canada (like in USA) I can't upload many GB to empty the EON . When the memory is full the openpilot dosen't work.
I don't have data plan on my cell phone with 50-100gb to do uploading.

Can we turn off recording ?? or delete manually clip on the EON to waive the uploading.

I think is a simply option to add on the next update.

bugsy924, 2018-05-07:
To delete
https://community.comma.ai/wiki/index.php/Configuring_OpenPilot
rm -rf /storage/self/primary/realdata/*

andy867, 2018-05-07:
If you join the comma.ai slack page, as listed on http://comma.ai

There are ideas on being able to offload the drive recordings to an external hard drive until you get somewhere where you can upload them via wifi.

DaveAL, 2018-05-07:
andy867 where is the slack page ?

bugsy924 there exist gitbash for Android (my phone) to do that without computer ?

but the option on the openpilot can be very nice

andy867, 2018-05-07:
Comma.slack.com

Will need to register an account. Once there, join #Comma, #openpilot, and #eon channels. Either @chase or @cody will be able to assist, (as well as bugsy) 😏

bugsy924, 2018-05-08:
Slack.comma.ai

closed 2018-12-08 by vanillagorillaa

smatterchoo, 2019-01-22:
@vanillagorillaa can I ask why this was closed?  will some release after 0.5.7 fix the issue of openpilot refusing to operate when storage is nearly full?   however you solve the issue (turn off dashcam recording, keep it on but when storage is nearly full just overwrite the oldest stored vids, compress / only keep the "interesting" bits a few seconds pre- and post- disengagement..) it seems anything is better than just having the device refuse to operate at all