commaai/openpilot #253: Add the ability to disable OpenPilot by software

mtourangeau, 2018-05-28:
I was think like clicking on the green steering wheel to disable OpenPilot and show a red steering wheel instead. Until I repress the streering wheel again, Open Pilot stay disable even if I activate the Toyota's cruise control. It can be enable by default when Open Pilot startup. 

This is to avoid removing the Giraffe when my wife want to drive and she wants use the standard cruise control radar on my Prius Prime Techno 2017. She doesn't want to fight against the steer when she drive.  I have to remove the giraffe for those moment. I don't like to connect/disconnect the camera all the time cause I'm wondering on a long terme, the camera's cable will break and will be hard to repair.

andy867, 2018-05-28:
Instead of removing the giraffe, you could always flip the dip switches to 1011 so that it restores stock system functionality as opposed to eon/openpilot.

mtourangeau, 2018-05-28:
I agree but I can't change the dip switches while she drive. The system is on and we won't stop for that.

mtourangeau, 2018-05-28:
I think 1011 is for Honda et 1001 is for Toyota by the way.

mtourangeau, 2018-05-28:
I wanted to solve this issue by my self but I can't find documentation on debuging python scripts on Windows. I'm not a linux fan.

mtourangeau, 2018-05-28:
Can you explain more in details proxy stock adas w/ panda?
https://github.com/commaai/neo/blob/master/giraffe/giraffe_sw.png

Is this to search for code when panda is conencted to a computer that can scann codes? I suppose, I'll need to use this to fix the Auto High Beam functionnality.

mtourangeau, 2018-05-28:
On the Tesla, Adaptative cruise control and LKA are different options. You can enable the cruise control without enabling the LKA.

andy867, 2018-05-28:
Are you on the comma slack channel? The comma team as well as python code gurus plus people familiar with panda, eon are on there (such as @erich Moraga, @vanillaagorilla, @j4z)

mtourangeau, 2018-05-28:
I am. I already asked my questions twice but no one answer so I gaved up.

mtourangeau, 2018-05-28:
I know that github is a source control but it would be easier if it would give the messenging possibility instead of writing comment here.

emmertex, 2018-09-21:
You can have a look at what has been done on Hyundai.
While its not quite what you want, if you rip the USB out of the EON while driving it will return to stock.

This is something we will continue to improve, but it will never be official.

mtourangeau, 2018-09-21:
You mean, if unplug, the Panda will pass the stock messages?

emmertex, 2018-09-21:
Yes, see this PR

https://github.com/emmertex/openpilot/pull/6

mtourangeau, 2018-10-25:
The best would be to use the LDA button on the Prius Prime steering to enable or disable OpenPilot without removing the stock cruise control. This button is disable when using openpilot but the perfect way would be like Tesla to have separated control. That would be awsome.

arne182, 2018-10-25:
This feature is available on my fork.
https://github.com/arne182/openpilot
With the addition of Joel Jacobs
https://github.com/arne182/DSUButtons
But a software button is also available for wife mode to be activated

mtourangeau, 2018-10-25:
Arn182, is it for the Prius?

arne182, 2018-10-25:
Originally for RAV4 but Prius should work as well. Give it a try and let me know.
cd /data && rm -rf openpilot* && git clone https://github.com/arne182/openpilot.git && cd openpilot && git checkout release2 && reboot

mtourangeau, 2018-10-25:
It's installed. I added my modification to remove the ping pong. I'm going to give it a try today.

wlee72, 2019-03-30:
> Originally for RAV4 but Prius should work as well. Give it a try and let me know.
> cd /data && rm -rf openpilot* && git clone https://github.com/arne182/openpilot.git && cd openpilot && git checkout release2 && reboot

Not sure why but this branch is not working on my 2017 Prius. The openpilot is not able to communicate with the car. Only CPU/BAT tempertures are displayed.

mtourangeau, 2019-03-30:
This branch is too old. You have to reflash your panda to make it work.

mtourangeau, 2019-03-30:
https://community.comma.ai/wiki/index.php/Panda_Flashing

mtourangeau, 2019-03-30:
Try a newer branch instead.

wlee72, 2019-03-30:
Will try later thank you