commaai/openpilot #255: Toyota Rav4 Road Sign System missing

arne182, 2018-05-30:
I have been trying to find the can bus messages for road sign recognition. 
Some how I assume it needs to be proxied from the forward camera. 
But need to find the messages that need to be forwarded first.
Anyone else missing this feature?

mtourangeau, 2018-05-30:
It wasn’t missing on my Canadian Prius Prime in the USA. But the sign does not works in Canada even without Open Pilot Installed.

arne182, 2018-10-25:
Done. Working. Don't know the signal yet.

zorrobyte, 2019-04-05:
The Prius Prime uses the same camera module, however this feature is disabled. Is it possible to spoof model encoding from the ECU to enable features on other models?

arne182, 2019-04-06:
This issue has been fixed with the forwarding of auto high beams via 120 ohm resistor

closed 2019-04-06 by arne182