commaai/openpilot #262: install openpilot on EON screen upside down

kmacpher67, 2018-06-11:
attempted upgrade of EON from chffrPlus to openpilot. 
1> uninstalled 
2> selected local WIFI end point
3> typed in:  openpilot.commai.ai
4> screen flashed old school characters and was inverted visually but screen touch was still the original way

kmacpher67, 2018-06-11:
https://photos.app.goo.gl/xArGsTrFGiumXGDp8

mtourangeau, 2018-06-11:
That’s normal.

kmacpher67, 2018-06-11:
It's unusable.  I can't even touch the spot on the screen where the text is. 
The tactile input is inverted from the display output.  That's normal? 
The system is unusable have you ever tried to figure out where to click when the output is upside down from the input?

energee, 2018-06-11:
This is a known bug, you can fix it from an ssh session with the following command:

`LD_LIBRARY_PATH="" content insert --uri content://settings/system --bind name:s:user_rotation --bind value:i:1`

jfrux, 2018-06-22:
Known bug, is there a way to fix it in code?  Just curious.

geohot, 2018-06-24:
0.4.7 should run that command automatically. We need to flip the installer too, will close this when we do.

geohot, 2018-06-25:
Fixed

closed 2018-06-25 by geohot

kmacpher67, 2018-07-18:
is `0.4.7` pushed to the `https://openpilot.comma.ai` latest? 
I just tried reloading: it's still upside down. 
---
was able to ssh into the system using this: 
https://medium.com/@jfrux/comma-eon-getting-connected-with-ssh-3ed6136e4a75
great doc, easy to follow. 
-----
```
total 0
root@localhost:/system/comma/home$ LD_LIBRARY_PATH="" content insert --uri content://settings/system --bind name:s:user_rotation --bind value:i:1
root@localhost:/system/comma/home$ 
```

kmacpher67, 2018-07-18:
Version 
0.5-release
Branch
release2

bugsy924, 2018-07-18:
0.5 is supposed to have the inward facing camera to the left of the screen

kmacpher67, 2018-07-18:
So my older model plastic case is nolonger valid.
There is a switch in the default geometry.
Thank you, @bugsy924 .