commaai/openpilot #273: visiond and gray / silver cars

jfrux, 2018-06-26:
Hey hey, I know visiond isn't actually OSS but I still wanted to report this here so it could be looked at:
But I had an incident with a grey/silver dodge neon… let’s just say I never thought something called a NEON would ever be camouflaged.   I’m assuming it’s `visiond` that would be responsible but it didn’t see it at all so it filtered it out.  Very strange behavior that I hadn’t seen until then.
Guessing it just blended into the roadway too much.

Figured I’d post it if you were interested in looking into the issue.

geohot, 2018-06-26:
visiond doesn't detect cars, radard does.

closed 2018-06-26 by geohot

jfrux, 2018-06-26:
@geohot Sorry about that.  Didn't realize.  Looking into that too then.