commaai/openpilot #278: Automatically delete oldest record when EON is full of storage

hailite, 2018-06-29:
when the storage is full, can OP delete the oldest record automatically to make it keep running?

mtourangeau, 2018-06-29:
You can ssh your eon with your phone and run this command:

rm -rf /storage/self/primary/realdata/*

I use the Terminal app on iPhone

onekrishnakumar, 2019-01-22:
> when the storage is full, can OP delete the oldest record automatically to make it keep running?

When I was on ver 5.6, this link worked
https://gist.github.com/chasebolt/fd5210b4c2a44a2b0db383162a66632c
As stated by Chasebolt this fork when installed does the following
"installs EON cron job that automatically purges oldest video when used space is greater than the MAX_USED_PERCENT limit"
This is exactly what it did, I tested it on ver 5.6
Then they forced v5.7 on us
But the above cron job does not work in v5.7
At least for me and I tried twice
So, for now, I SSH into my eon and run the following
rm -rf /storage/self/primary/realdata/*
This basically removes all the drive recordings
The memory goes back to 3% 

For a more permanent fix, Chasebolt would have to update the above code for the openpilot v0.5.7 release

If you know of a better way, please share

mikecboyd, 2019-01-27:
Seems long term solution is OP GUI update in Settings to "Erase Previous Drive Recordings" by incorporating above script after some review.

vanillagorillaa, 2019-05-20:
Resolution in 3f9059fea886f1fa3b0c19a62a981d891dcc84eb

closed 2019-05-20 by vanillagorillaa