commaai/openpilot #287: Lexus IS Series Compatibility

kushal-jaisingh, 2018-07-12:
I noticed that none of the Lexus IS series vehicles are included on the compatibility list on the github readme. However, it does have adaptive cruise control and lane keep assist. That being said, is openpilot currently compatible with the 2018 Lexus IS-350? If so, any reviews how well it works? I was searching for some video reviews of openpilot and most of them were great. However, out of all the reviews I managed to find, there was only 1 review for a Lexus vehicle (the RX series), and in the video they noted that for some reason the car would keep hugging the right shoulder of the road. That being said, has this issue been resolved and how is the compatibility with other Lexus vehicles specifically the 2018 IS-350? Thanks

thomaspich, 2019-08-05:
Hi @kushal-jaisingh ! It must be compatible, I have ported OpenPilot on a 2018 Lexus (CT200h)

quillford, 2019-08-05:
My [fork](https://github.com/quillford/openpilot) should work but might need to change the fingerprint. Join the discord if you need help. https://discord.comma.ai/

Mason0916, 2019-08-26:
> 嗨@ kushal-jaisingh！它必須兼容，我已經在2018年雷克薩斯（CT200h）上移植了OpenPilot

Hi @thomaspich  CT200H Can it be used normally?

eFiniLan, 2019-08-26:
@Mason0916 CT200H 已經有 PR 給官版了，在沒整合進官版前可以找第三方的版本安裝

thomaspich, 2019-08-26:
Hi @Mason0916, I made a pull request on comma's branch, normally this should be merged in the next releases. If you want to have Openpilot on your CT200h now,  just clone my repo https://github.com/thomaspich/openpilot and git checkout on  **lexus-ct200h-port** branch

thomaspich, 2019-08-26:
@Mason0916 When you don't have disconnected your DSU (which does ACC) you can enable OpenPilot only above 45kph, and it stops at 40kph. The LDA works until 40kph too, because OpenPilot does it.
Have you see how to install the EON ? It's pretty easy ! Just disconnect the FRC cable and plug the giraffe in the middle
Are you on Discord ? It's easier to talk on it with my phone

Mason0916, 2019-08-26:
> 嗨@ Mason0916，我在逗號的分支上發出了拉取請求，通常這應該在下一個版本中合併。如果你想在CT200h上安裝Openpilot，只需克隆我的repo https://github.com/thomaspich/openpilot和git checkout on **lexus-ct200h-port**分支

I am a Ct200H owner, the original ACC can only start at speed above 45, Openpilot can follow the car to speed 0?  The original LDA is also used at speeds of more than 50 kilometers per hour. Is it necessary to dismantle the original computer?

rbiasini, 2019-09-11:
Lexus IS is fully supported since 0.6.4 openpilot release.

closed 2019-09-11 by rbiasini