commaai/openpilot #300: Steer oscillation for Prius/Prime

mtourangeau, 2018-07-17:
This is to track the steer oscillation for the Prius/Prime which I think is a knowed issue.

I need someone else to test steer steer oscillation fix for a Prius/Prime. I edited the selfdrive/car/toyota/interface.py and changed 2 values. I need someone else to test those value and give me more feedback. 

The originales values are: (selfdrive/car/toyota/interface.py under if candidate == CAR.PRIUS: line 81)
ret.steerKpV, ret.steerKiV = [[0.4], [0.01]]

The steerKpV is to center the car in lane. I lowered the value to 0.3 which keep the car a little more on the left cause I finded too near to the right lane. 

The steerKiV I’m not sure exactly what it is but I review a changes for the rav4 (Fixes wobble on new CRV #269) that help a lot the oscillation by lowering this value so I tried the same thing with my Prius Prime. I setted it to 0.005. 
ret.steerKpV, ret.steerKiV = [[0.3], [0.005]]
I did not play a lot with this one because I didn't know exactly what it does but the oscillation is much much better that the original value. In the next few days, I will try different value. If someone can explain more in detail, what exaclty this variable does, it could help me. What are the min and max value?

Can someone else give it a try?

oliverrevol, 2018-07-20:
May kinda PID algorithm to help smooth the steering. JUST GUESSING...

REF.
https://en.wikipedia.org/wiki/PID_controller
The proportional term produces an output value that is proportional to the current error value. The proportional response can be adjusted by multiplying the error by a constant Kp, called the proportional gain constant.
Response of PV to step change of SP vs time, for three values of Ki (Kp and Kd held constant)
The contribution from the integral term is proportional to both the magnitude of the error and the duration of the error.

ErichMoraga, 2018-07-24:
I wasn't aware someone had a PR for "Steer oscillation for Prius/Prime".  One should be aware I've got my own branches I've been sorting this out with, and of course my PR for Prime differentiation was already approved.  You might as well join #erichmoraga in Slack too.

Ping Pong in straights has been mostly eliminated for months, so I've been sorting it out for curves.  I really don't recommend approving any PR like this 'til there's major consensus among the Prius Prime group, one of the reasons my channel is mostly a select group of Prime drivers.

mtourangeau, 2018-07-31:
I leave my latest number here for reference or in case I download a newer version and overwrite my value on my eon. So it’s kind of a backup 😀

81       ret.steerKpV, ret.steerKiV = [[0.35], [0.    0002]]
 82       ret.steerKf = 0.000075   # full torque fo    r 10 deg at 80mph means 0.00007818594
 83       # TODO: Prius seem to have very laggy act    uators. Understand if it is lag or hysteresis
 84       ret.steerActuatorDelay = 0.1

legonigel, 2019-04-09:
Give the age of this issue and recent changes in controls. If the issue still affects the latest release, it can be reopened or a new issue opened.

closed 2019-04-09 by legonigel

mtourangeau, 2019-04-09:
It’s not fixed but a work in progress.

ErichMoraga, 2019-04-09:
I basically have it fixed with some code I've been testing.  
@mtourangeau can DM me on Discord for details if he wants to test.

pratikchhatbar, 2019-05-08:
@ErichMoraga I would like to test the code - I've been using EON 0.5.10 on my Prius Four 2016 and pretty much this lateral wobble fix would make it an order of magnitude better!