commaai/openpilot #306: Dodge Charger Support

FallenDev, 2018-07-20:
I'm extremely interested in this and have tinkered with my chargers computer in the past using "Tazer" Which unlocks a ton of system features. Dodge Charger has adaptive cruise and lane keep assist with steering assist. It also has a computer that's been opened up which has a ton of things not locked down. I'd like to see where I can go with this.

adhintz, 2018-07-20:
I have a partially working port for [Chrysler Pacifica here](https://github.com/adhintz/openpilot). Here's my information on the [Pacifica with Cabana links and canbus messages](https://github.com/adhintz/opendbc/wiki).

Chrysler and Dodge are part of the same company, so it's possible that their canbus messages are very similar. If you have a Panda (from shop.comma.ai), then feel free to share a few Cabana links of your drives and we could see how similar the canbus messages are.

giusemc1988, 2018-07-25:
I have been researching this for dodge charger chrysler 300 for so long, Im not a coder but I really like this openpilot work. My chrysler 300 has electro-hydraulic steering, but I have install full electric rack and pinion from 2015-up chrysler 300, and it works. Now, I hope someone could help me make openpilot work . I have found some guy make it work but I dont really know how he did it. Here down bellow is his youtube link. Hopefully, someone here could help me do it
https://www.youtube.com/channel/UCr8FnMEqf55vP8WbWcUmwAA

FallenDev, 2018-07-25:
Thank you very much for your reply, I'm going to take more interest in this and do some DD.

adhintz, 2018-07-25:
@giusemc1988 Thanks for the link to the video for the Chrysler 300! I'll try to find the person that did it and see if they'll share some information.

Last night I got my Chrysler Pacifica working well with OP: https://www.youtube.com/watch?v=9eY1gPay7ao

If you have a Panda, feel free to share some Cabana links with me and I'll see how similar the canbus messages are.

giusemc1988, 2018-07-25:
I would love to do anything that can help u finger out how to make it work on Chrysler 300. But i am not programmer, but im mechanic, so do u think i can help with anything?

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/commaai/openpilot/issues/306#issuecomment-407856758",
"url": "https://github.com/commaai/openpilot/issues/306#issuecomment-407856758",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [commaai/openpilot] Dodge Charger Support (#306)",
"sections": [
{
"text": "",
"activityTitle": "**Drew Hintz**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@adhintz",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"commaai/openpilot\",\n\"issueId\": 306,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"commaai/openpilot\",\n\"issueId\": 306\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/commaai/openpilot/issues/306#issuecomment-407856758"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 358641117\n}"
}
],
"themeColor": "26292E"
}
]

giusemc1988, 2018-07-26:
Thank you for all of your replies 

I would love to do anything that can help u finger out how to make it work on Chrysler 300. But i am not programmer, but im mechanic, so do u think i can help with anything?

vanillagorillaa, 2018-12-18:
Seems no progress here, suggest to move it to slack for any more discussions.

closed 2018-12-18 by vanillagorillaa