commaai/openpilot #311: Corolla support

atxfz09, 2018-07-31:
I am having issues with my giraffe and panda not supplying power to the neo, causing it not to work

mtourangeau, 2018-07-31:
Follow this procedure: https://community.comma.ai/wiki/index.php/Panda_Flashing

The easiest way to flash panda via EON is as follows: 
Disconnect Panda from Giraffe
Plug Panda into OBD2 Port of car (Car powered off)
Start with EON powered off and connect via USB to Panda
Once EON powers up the Panda will begin flashing and green LED will flash for a few seoncds
Flashing is complete when Panda shows red pulsing LED (heartbeat)
Plug Panda back into Giraffe

mtourangeau, 2018-07-31:
you can use your phone camera to check the flashing light

atxfz09, 2018-07-31:
I got the panda to flash but now I keep getting a cruise malfunction warning

mtourangeau, 2018-07-31:
Ask your question in the community forums. 

https://community.comma.ai/viewforum.php?f=19&sid=4047bd9da6f83dfcebd2630915ff825b

rbiasini, 2018-07-31:
And join comma.slack.com!

closed 2018-08-16 by rbiasini