commaai/openpilot #339: Did we stop doing release tags?

jfrux, 2018-08-28:
The reason I ask is because I'm using GitHub API to track releases and things on https://database.opc.ai/ and I'm not seeing 5.1 or 5.2 so I checked here and they do appear to be missing.

Just curious...

ch4se, 2018-09-01:
Looks like they missed adding the release tags since 5.0

jfrux, 2018-09-13:
No? Nothing? Still trying to ensure I'm properly tracking versions for opc.ai.
I want to ensure that Guides, Comments, etc, are relative to the version at that time and allow us to maintain what guides still are relevant to particular versions but I want to also know what versions have been pushed to release2.

Tags is generally how this is done but tags are missing since 0.5.

jfrux, 2018-09-21:
Nope, cool... just was curious! Thx for the info. 🥇🤣❤️

energee, 2018-09-21:
If you want tags for opdb, energee repo has release tags 👍

kmacpher67, 2018-10-06:
Why doesn't the main comma.ai repo have Release tags/branches anymore?
Isn't there a release build script that does this automatically?
Are the test failing and the release tag is not getting cut?
Idk enough about this project pipeline process, but who is that controls those decisions?
Is there an architecture or devops review committee/team for these things?

jfrux, 2018-10-06:
lol

vanillagorillaa, 2018-12-18:
Release Tags are back! all releases in release2 are tagged and will be tagged from now on with each release!

closed 2018-12-18 by vanillagorillaa

jfrux, 2018-12-18:
It’s so sexy when you talk taggy to me.