commaai/openpilot #358: Have you thought about to support BYD?

vell001, 2018-09-08:
BYD is a famous vehicle manufacturing company in China, they released the open API just a few days ago, see: [http://oip.byd.com/sites/developer/index.html](http://oip.byd.com/sites/developer/index.html). That's the API document: 
[比亚迪智慧开放平台API说明书V1.0.5.pdf](https://github.com/commaai/openpilot/files/2363166/API.V1.0.5.pdf)
We can get the whole vehicle information easily by the open API, but we can not operate the vehicle's steering wheel and throttle right now. There may be opportunities here, because of BYD's open mind. Maybe it's easier for us to support BYD cars, right?

muyufan, 2018-09-09:
That api is useless, if the car has ACC and LKAS and use CAN, then any car may be supported.

closed 2018-12-18 by vanillagorillaa

pascaldls, 2023-10-27:
2023 BYD Atto 3 models are currently equipped with advanced driver assistance systems (ADAS) including autonomous emergency braking (AEB), lane-keep assist, and adaptive cruise control with stop/go

May be it could be reviewed

ccckblaze, 2024-02-28:
https://github.com/BYDcar/opendbc-byd

someone make some effort, but still no further break through

ccckblaze, 2024-02-29:
a byd atto 3 fork was made
https://github.com/commaai/openpilot/discussions/28126

wenchaoliu-93, 2025-04-20:
I already found a product listing on Taobao that claims BYD support. Records show they have sold over three hundred units, and that they have over two hundred reviews.

Link is [here](https://item.taobao.com/item.htm?from=cart&id=773413540122&pisk=gVxobzAN1U75XflAknsSSe7OGm3YVgsCTBEd9MCEus5X2Y_d9EXe9IMIFgpFtBAAG3H5J6L0KdODtwM5965H9p8-Hcn9FLsCYMctXcBrBa7BZ9zPYv5VQOkAUUdeJ1ICYfh3kk7SjMOyxzZa8rWVG9XUaMRF0iWCEMWP8_kcg9X_87ReTxbVp92zUwrr0sWNg7PPa6SVgOBNTT7F8xvVG9SFTB7rnKlDvWCM8ExqG3q-CyBdTnXlr1JmlkrH9lQP_ef4Yf-cEf1wUsqUY_HpSVJDebq6HgtBsOd-0uAMCnOc3ConiMOX8BWHgcNdVFdWfZA-7X9c09bNYZ2UYKjPZwW5bcPchEJJ83OuKkWW2hQh1ZDUAwI2XN8wZJM63gWeONtK1u1yLdt6WGmYzGJD8CYh4Ua478y4Atkp0yaCzt6c69ORIx4p6xTqnx4_naWfFAHmnyaCzt6c6xD07obPhTMO.&skuId=5758307511349&spm=a1z0d.6639537%2F202410.item.d773413540122.61677484ZzLxjU).