commaai/openpilot #370: [Feature] Communication Between Vehicles

asddsaz, 2018-09-20:
**Feature**: In order to provide a more secure driving experience between Comma AI users and forks, I propose the motion to add op-in mesh network capabilities. This would help with possible glare, partial/full covering of cameras/sensors, dangerous intersections, u-turns, and unexpected change in events. 

**How**: I propose using IOTA or NANO and/or Mesh Networks for communicating.

NANO: https://nano.org/

IOTA: https://www.iota.org/

geohot, 2018-09-25:
I very much support this motion. In fact, if it can handle the camera being covered, why do we have to have a camera at all? I've also heard synapse.ai with mesh networking can decentralize the AI, so it sounds like without the AI we won't need an expensive processor. We can even use Raspberry PI!

Looking forward to your pull request and the money we will save on next generation cameraless Raspberry PI EON powered by blockchain technology!

jfrux, 2018-09-25:
Bahahahah nailed it.

closed 2019-09-11 by rbiasini