commaai/openpilot #382: How to create a whole system for Chevy Volt 2017?

tom121212, 2018-09-30:
My car don't have ACC. So, I only need the board and cable from camera to phone, right?
Now, where can I bought that board?

vntarasov, 2018-10-01:
Currently only ACC-capable GM cars are supported. Otherwise, you can only use EON & chffrplus in read-only mode. Reach out to folks in #gm channel on slack.comma.ai if you have more questions.

tom121212, 2018-10-03:
If, I mean if, I bought an EON system and connect it in my chevy volt 2017. What would happen? System crash?
Still able to drive, but EON system not work?
Or, EON can control my wheel. My car can drive automaticaly in the lane center?
Which one?

kmacpher67, 2018-10-06:
I have a Chevy Malibu that I have hacked the profile to capture the canbus fingerprint of the Malibu and map it to the Chevy volt profile.
At startup somewhere magical the system is trick into thinking my car is a Chevy volt but because I don't have the codes for the LKAS the eon boots in ChffrPlus or read only mode.
If your vehicle's ID fingerprint matches the standard Chevy volt then && you have LKAS then likely it would boot into open pilot mode.
The self driving stuff doesn't kick in until you enable cruise control.
Forking the repo and changing the default Chevy voltsb fingerprint is somewhat trivial.
Creating all the hooks to get another profile enabled is more non trivial.

closed 2018-12-12 by vanillagorillaa