commaai/openpilot #393: Prius: Do not disengage openpilot when pressing the gas pedal

mtourangeau, 2018-10-12:
Touching the gas pedale should not disengage the Adaptative Cruise Control with the Toyota Prius/Prime.

energee, 2018-10-12:
This is part of openpilot safety requirements:

> The driver must always be capable to immediately retake manual control of the vehicle, by stepping on either pedal or by pressing the cancel button.

See https://github.com/commaai/openpilot/blob/devel/SAFETY.md

mtourangeau, 2018-10-12:
Then disable open pilot but not the Adaptative Cruise Control. I never had in my life a Cruise Control that disable the ACC by pressing the gas pedal.

mtourangeau, 2018-10-12:
"With the stock Driving Support Unit (DSU) connected (or in DSU-less models like Camry and C-HR), the acceleration is controlled by the stock system and is subject to the stock adaptive cruise control limits. "

The stock adaptative cruise control will not disangage because you use the gas pedal.

fkong777, 2018-10-18:
I would agree with mtourangeau.  OEM toyota system do not disengaged DRCC by pressing gas pedal.  it will be so helpful in stop and go traffic.  without pressing Resume button all the time.

jfrux, 2018-10-18:
How does pressing the gas benefit you in stopped traffic?!? Is it because of the minimum engage speed? When your car has real stop and go, like when you add Comma Pedal, you literally never disengage because you never need gas or brake the entire drive.
I think that any code to change gas disengage is not helpful to the end game and just creates further safety issues.

Yes it’s annoying when you don’t have actual low speed follow / stop and go but get Pedal and you’ll have it. :)

Just my two cents.

mtourangeau, 2018-10-18:
Basically when the traffic stop completely the car stop. Then you have to press the gas pedal to continue or press the resume button. It would be nice if open pilot can continue to control or resume by it self. 

Also, when you in fast traffic like 45-55MPH, and you want to by’ass one one or what ever then you use the gas pedal and open pilot stop. 

It’s a useless discussion since OP will never do it. I’m going to delete this issue.

jfrux, 2018-10-18:
I just think it seems like more of a paradigm shift that you’re having to get accustomed to than it is anything else…
After pressing the gas, OP disengages since you’re no longer self-driving you’re asking to take over.
At that point, simply press the button again to continue your journey… its as simple as that.
Also, OP doesn’t fully engage until you let off the gas after engaging OP.

TLDR; why is it so hard to just press the silly button again to set your new speed to the speed you accelerated to?
Seems like a silly problem to “fix” considering the massive disadvantages it causes from feeding bad data about acceleration.

This is the Tesla model which is pretty much I think the way the future of autonomous driving looks like.

You’re either in “openpilot mode” or you’re not.  One or the other.
The only reason steering doesn’t disengage is because it is useful for Machine Learning and fixing mistakes.
Staying engaged while accelerating messes with the data I’d imagine and is no longer self-driving.

We’re just having a discussion… its a discussion that is worth having as others have also complained about this.
But again, my two cents are that its just a matter of getting used to driving with OP versus with the Stock systems.
The stock systems are bandaids put in place by the car companies for various reasons but their intentions aren’t the same as OP.
OP appears to have more of the Tesla model which is the end game is, we shouldn’t need to touch anything.  Where as, the stock system that so many seem to love requires you to basically be tethered to the steering wheel AS MUCH AS POSSIBLE so that your family doesn’t sue the car company when you inevitably die from not driving your own car with the shitty stock system that is presently in the majority of our cars.

It’s a trade-off for sure with Openpilot… but I think it’s the way to go long term. “BIG PICTURE” type of view.

Healthy discussions.  But in the end, Comma needs the data in a clean manner to build accurate maps and planning code.
Without that data, OP doesn’t exist and stays the same old LKAS/ACC features you get from your stock system.
THE ENTIRE PURPOSE of using Openpilot versus your stock system is OTA Updates, Better Vision Model, Longer engagements without having to do anything.
It’s NOT its purpose to just replicate your stock features all together.
And I’m eternally grateful for that, as the majority of stock systems put in cars are currently extremely limiting and will be for many years to come since most car companies aren’t yet willing to roll out something like what Tesla has.

<3

jfrux, 2018-10-18:
Also, should be noted that much of what I just said is based on my limited understanding of Openpilot and Comma’s overall intentions from what I’ve seen and experienced.
It is in no way an official response or meant to be definitive.  Just talking here folks.

mtourangeau, 2018-10-18:
I trust more my stock breaking system then open pilot breaking system. For me, OP drive and should’t care about breaking and accelerating since that the stock system that do this functinality.

mtourangeau, 2018-10-18:
Close this issue.

closed 2018-10-18 by mtourangeau

zorrobyte, 2019-04-11:
@mtourangeau @fkong777 see this commit for the possibility of stop and go on Toyota without user input, it works on my Prime (be sure to read comments): https://github.com/zorrobyte/openpilot/commit/4b70e41e9b87c6a01f7fff3f018ab1739e0038f2

cydia2020, 2019-10-11:
@zorrobyte the commit works, but it will use a LOT of battery as the car never truly "stops", just give the cruise leveler a wiggle instead