commaai/openpilot #398: 2016 Honda Civic doesn't identify car cutting off from right

KyleBoyer, 2018-10-18:
Hey guys,

Yesterday while driving on the highway during rush hour, a car 'cut' me off in traffic, kinda slowly, but OP did not recognize it in the lane until it was half way over, even though the video you can clearly see it go over the lines. I'm assuming this is because OP is using my radar data to aid in detecting the lead car, so is it possible to aggregate this data to accomodate for the Civic's radar sensor being on the driver side of the car?

Video of what happened:
https://youtu.be/5YMQApmZOsg

jfrux, 2018-11-07:
It's 2018, it can identify as whatever it wants!

arne182, 2018-11-07:
Were you not using my 0.9s following distance mod?

KyleBoyer, 2018-11-07:
@arne182 no when I took this it was on stock OP before I ever even installed a different fork of OP.