commaai/openpilot #447: Compiling ui

brubakerjeff, 2018-11-21:
Hi

I am trying to run make inside of UI and receive the following. Am I missing a dependency?

$ make
[ CC ] ui.o
clang -std=gnu11 -g -fPIC -O2 -Werror=implicit-function-declaration -Werror=incompatible-pointer-types -Werror=int-conversion -Werror=return-type -Werror=format-extra-args -DQCOM -MMD \
          -Iinclude -I.. -I../.. \
          -I../../phonelibs/nanovg \
          -I../../phonelibs/zmq/aarch64/include \
          -I../../phonelibs/capnp-c/include \
          -I../../phonelibs/json/src \
          -I../../phonelibs/opencl/include \
          -I../../phonelibs/linux/include \
          -c -o 'ui.o' 'ui.c'
ui.c:9:10: fatal error: 'cutils/properties.h' file not found
#include <cutils/properties.h>
         ^~~~~~~~~~~~~~~~~~~~~
1 error generated.
make: *** [Makefile:95: ui.o] Error 1
$

EternityLan, 2018-11-22:
try ```./start.sh`` instead?

brubakerjeff, 2018-11-22:
Same error. I am using “standard” lineageos. Do headers have to be extracted from a lower layer (inside the build source code) and included with the build code to allow UI to run? Where is cutils on the custom os?

brubakerjeff, 2018-11-25:
I'm thinking the needed files are in https://github.com/commaai/usr/tree/master/include/cutils. Now need to find a package management way to install.

legonigel, 2019-04-09:
This doesn't seem to be an openpilot issue. Closing.

closed 2019-04-09 by legonigel