commaai/openpilot #460: Chffr app won´t record in background

nirodg, 2018-12-10:
Recording in background work as expected if, firstly you have Chffr opened and ready to record, then when starts recording you switch to a different app, keeps recording. 

But it wont work if, before Chffr starts recording, you switch to a different app. If will begin recoding once you switch back to Chffr.

Device SM-G960F Android 8.0.0 - Samsusung Experience 9.0
chffr 2.2.3.88

omerjerk, 2019-02-05:
Is the source code of this app open by any chance? I can look into this issue.

andy867, 2019-02-05:
The reason that chffr app won't record in the background, at least per apple terms for the ios app, is that Apple requires any still imagery/videography to be the active application so if running said app and switch to another app, it stops recording as it's no longer the active application.

legonigel, 2019-04-09:
Closing as wontfix.  The app will not be open source

closed 2019-04-09 by legonigel