commaai/openpilot #474: Evidence of higher steering torque for Honda's

Gernby, 2018-12-19:
After speaking with @rbiasini yesterday about recent observations in my Accord, I performed a simple test and captured a datalog ([here](https://community.comma.ai/cabana/?route=ddd3e089e7bbe0fc|2018-12-19--11-50-46)).  In the log, from 0:42 - 0:46 the stock system overrides OpenPilot's steering commands somehow, and generates a much stronger steering result.  During that time, the steer_torque_motor parameter becomes much more significant, and its phase shifts ahead of the steer_torque_sensor parameter.  The steer_angle_rate also becomes much larger.

I suspect that there is more than 1 data stream that the EPS responds to, with different torque limits.  I think that Road Departure Mitigation might be the system in control here, since my HUD was flashing yellow lane lines.  
[image]

There are lots of unmapped frames from the camera, but 0xE5 seems related.
[image]

adhintz, 2018-12-19:
Thanks for digging in to this!

The Cabana link you posted is only accessible by yourself. If you want everyone to be able to see it, when viewing the drive in Cabana, click "Copy Share Link" and then paste the URL from your clipboard. The URL should be much longer and have a url parameter in it.

csouers, 2018-12-19:
e5 is indeed related (for Bosch only it seems). From my research, it appears to only control the wheel wobble during a LDW/Correction event. The radar combines e4 and e5 from the camera into a single e4 frame that gets sent to the EPS. (Look at e4 on bus 0)

On civic, the wobble occurs while e5 is active and the corrective torque is applied through e4 immediately after.

Regardless, we could make the wheel wobble part of the driver monitoring inattentive alert. I am experimenting with how to get wheel wobble and OP command torque, but without the warnings on the hud.

Gernby, 2018-12-20:
> re Link" and then paste the URL from

Big oops ... I've updated the cabana link...  Thanks for pointing it out!

Gernby, 2018-12-20:
That is really good to know about the combination of the messages, but I think there's more to it than just the wobble.  The wobble seems like a first stage (Lane Departure Warning), but the next stage is much more impact-full and harder to reproduce.  I wouldn't necessarily consider this particular event to be second stage, but I do have a few cabana drives that capture it.  I've shared them on Slack, but they are long gone from history.  I'll see if I can dig them up again.

Gernby, 2018-12-20:
I found the most significant example(s) of apparent RDM intervention that I've ever experienced ([here](https://community.comma.ai/cabana/?route=ddd3e089e7bbe0fc%7C2018-12-05--09-06-38&max=26&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2Fddd3e089e7bbe0fc%2Fccedf3ffcd92c1e2147842dcdae8ae5d_2018-12-05--09-06-38)).  There were several episodes over a few minutes between 16:00 - 22:00 while I was on the highway.  Traffic was about as high as it could be at this speed.  The sun was in a very non-optimal position, and the lane markings on the highway suck.

[image]

This was the most intense episode, which resulted in a very fast 16+ degree adjustment in steering before I could grab the wheel.
[image]

Gernby, 2018-12-20:
I was able to capture better data ([here](https://community.comma.ai/cabana/?route=ddd3e089e7bbe0fc%7C2018-12-20--10-13-26&max=10&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2Fddd3e089e7bbe0fc%2Fa1e06412f88751a56d427debc63a367a_2018-12-20--10-13-26)) by putting OpenPilot into Passive mode, so that I could observe the stock RDM behavior more clearly.  This is with LKAS disengaged, but with the RDM set to "Narrow".  

During this drive, I found that the steering system did "wobble" at first, then applied significant steering force back toward my lane.  There's a clear relationship between e4 and e5, and it appears that some bits currently defined as SET_ME_X00 and an undefined bit may change the EPS gain or mode.

Below is a screenshot of my dashboard, showing the binary view of 0xe4 and 0xe5 during a couple RDM engagements.  Note that the data has been down-sampled to 10 frames per second.

[image]

Gernby, 2018-12-20:
From cabana
[image]

csouers, 2018-12-20:
Which e4 are you looking at? CAN 0 or 1/2?

Gernby, 2019-01-29:
After more testing, It seems that the EPS isn't really torque limited as much as rate limited.  While testing at low speed in a parking lot, steering with a keyboard, I found that the EPS would steer quickly for a few degrees, then slow down for a few seconds, then steer quickly again for a few degrees.  This was a repeatable behavior in both directions.  It seems that the steering "bursts" are about 5-6 seconds apart, without much relation to the value of the steering commands.  In the screenshots below, the steering commands (via keyboard) were multiple times the standard maximum value (0x1000).

[image]
[image]

I've also observed somewhat similar results while testing other feed-forward strategies.  I'm not sure how I would interpret this result, but it seems to indicate that the EPS has a max degrees per second that it will perform.

[image]

The reason why I think this may relate to my occasional "big steering torque" events, is because I think that the EPS may not limit itself so much when some other bit or bit combination is activated on the bus.

geohot, 2020-01-23:
What's the status on this? Can higher steering torque be reproduced?

gregjhogan, 2020-01-24:
The evidence in the original post is completely bogus.  The signal identified as `STEER_TORQUE_MOTOR` is actually identical to `STEER_ANGLE_RATE` inverted.  The rest of the data presented does not show anything conclusive and probably doesn't belong in a github issue.  Feel free to discuss further in discord and submit a PR if you find something that works!

closed 2020-01-24 by gregjhogan

Gernby, 2022-01-29:
> The evidence in the original post is completely bogus. The signal identified as `STEER_TORQUE_MOTOR` is actually identical to `STEER_ANGLE_RATE` inverted. The rest of the data presented does not [...]

The data provided in the original post was not "bogus", but the DBC at the time did have incorrect labels for some signals (including STEER_TORQUE_MOTOR).  

I'm circling back to this issue now, just to see if I can make any sense of the behavior, since I still believe there is a higher level of torque available for request by VSA.  Techinfo describes several VSA parameters that the EPS uses.

gregjhogan, 2022-01-30:
I specifically said the evidence in the original post was bogus, sure the data was real, but it was clearly interpreted incorrectly (since it was mislabeled and it is not torque) and this was specifically mentioned as the evidence.

Please use other places such as discord to discuss your theories, and I look forward to a PR if you find something useful!