commaai/openpilot #476: EPS Faulting since NEOS v8 update - 2018 GMC Acadia Denali

CryptoKylan, 2018-12-20:
i really hope someone can help me.  For whatever reason since i updated my eon to neos 8 iv been having issues where i get an EPS faults randomly.  every time this happens i see theres a late/early ASCMLKASteeringCmd msg.  it seems as if when the EPS receives two ASCMLKASteeringCmd msgs in under 0.010ms it faults out.

normally the msgs should be about 0.020ms apart or a steer step of 2 as per stock.

the ASCMLKASteeringCmd msg timing is the only common denominator i can find on every fault.  This happens engaged or not.  i tried with different usb cords and another panda.  everything has been fine for months, the only thing that changed recently was the neos update but im not sure how that could have changed anything.

this is on the comma devel branch too.  i even tried factory resetting my eon just to be sure and i still experience this problem.

i had a thought to try and test this timing theory,  maybe if i tried delaying the ASCMLKASteeringCmd  msg to a step of 3 or about 0.030ms it would have less of a chance of having a delay and causing the following msg to be sent under 0.010ms from the previous lka msg.  

by doing this i was not able to reproduce the EPS issue.
heres the drive from that testing.  this is just one of many tests.
https://community.comma.ai/cabana/?route=7cc2a8365b4dd8a9%7C2018-12-19--13-16-22&max=48&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F7cc2a8365b4dd8a9%2F87d09f383607dd19cb4a21a9cb9b6fbc_2018-12-19--13-16-22

So to try and test this theory further i set the steer step to 1 which would be sending the ASCMLKASteeringCmd msg every 0.01ms which should pretty much guarantee a msg in under 0.01ms and therefor should fault out almost right away..  

sure enough the EPS does fault out pretty much right away every time it receives 2 ASCMLKASteeringCmd msgs.
https://community.comma.ai/cabana/?route=7cc2a8365b4dd8a9%7C2018-12-19--17-29-59&max=41&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F7cc2a8365b4dd8a9%2F54904d53333f5b094f6211e8f705d254_2018-12-19--17-29-59

im not sure what i can do to try and resolve this or where to even start.  

is there a way i can make sure the ASCMLKASteeringCmd msg is never sent any sooner than 0.010ms after the previous msg?

i used a screen recorder and could see the controlsd lag build up a bit at the time i get the fault so im not sure if that has anything to do with it

--------------------------------------------------------------

problem drives

12:55:25 - 12:58:25 - seekTime=894
parked after drive - late/early ASCMLKASteeringCmd msg? - 894.251 - 894.259
https://community.comma.ai/cabana/?route=7cc2a8365b4dd8a9%7C2018-12-18--12-43-25&max=15&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F7cc2a8365b4dd8a9%2F0aaa6f910b9885b07bf3c563084427e9_2018-12-18--12-43-25

20:01:05 - 20:04:05 - seekTime=604
late/early ASCMLKASteeringCmd msg? - 604.490 - 604.529
https://community.comma.ai/cabana/?route=7cc2a8365b4dd8a9%7C2018-12-17--19-52-05&max=29&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F7cc2a8365b4dd8a9%2Ffac15d8847e972be0af85395da5fbdf2_2018-12-17--19-52-05

14:57:09 - 15:00:09 - seekTime=544
late/early ASCMLKASteeringCmd msg? - 544.890 - 544.930
https://community.comma.ai/cabana/?route=7cc2a8365b4dd8a9%7C2018-12-17--14-49-09&max=28&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F7cc2a8365b4dd8a9%2Ffcda4af77cef4657c7a55bfbf7bf5697_2018-12-17--14-49-09

13:49:51 - 13:52:51 - seekTime=128
large amount of dropped ASCMLKASteeringCmd msgs 128.494 - 128.801
https://community.comma.ai/cabana/?route=7cc2a8365b4dd8a9%7C2018-12-17--13-49-51&max=8&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F7cc2a8365b4dd8a9%2Fc5a9d6b472c6e6546c8d8909647959ca_2018-12-17--13-49-51

13:12:13 - 13:15:13 - seekTime=623
ASCMLKASteeringCmd extra msg? 623.912 - 623.918
https://community.comma.ai/cabana/?route=7cc2a8365b4dd8a9%7C2018-12-17--13-04-13&max=13&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F7cc2a8365b4dd8a9%2Fea58a0e2a094b72577bb31a270365803_2018-12-17--13-04-13

18:59:59 - 19:02:59 - seekTime=1873
not engaged.  late ASCMLKASteeringCmd becomes 2 msgs at the exact same time @ 1873.915
https://community.comma.ai/cabana/?route=7cc2a8365b4dd8a9%7C2018-12-15--18-30-59&max=36&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F7cc2a8365b4dd8a9%2F371205592513649334672473064953af_2018-12-15--18-30-59

16:26:13 - 16:29:13 - seekTime=656
late ASCMLKASteeringCmd becomes 2 msgs at the exact same time @ 656.661
https://community.comma.ai/cabana/?route=7cc2a8365b4dd8a9%7C2018-12-15--16-17-13&max=14&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F7cc2a8365b4dd8a9%2F16af576e14e34c5be1a1115a5e58565e_2018-12-15--16-17-13

----------------------------------------------------------------------------------

stock drive / stock lkas - 18:03 - 18:06
https://community.comma.ai/cabana/?route=7cc2a8365b4dd8a9%7C2018-09-12--17-57-31&max=52&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F7cc2a8365b4dd8a9%2Fb144fb331c158412cdc93ec42cb059eb_2018-09-12--17-57-31

-----------------------------------------------------------------------------------
-----------------------------------------------------------------------------------

Pictures showing the issue

[image]
[image]
[image]
[image]
[image]
[image]
[image]

mtourangeau, 2018-12-20:
I’m having the same kind of issue with my Prime 2017. Seems that if the car is already started when the EON boot up, fix this problem.

mtourangeau, 2018-12-20:
I have the same kind of issue with my Prime 2017. I found that booting the oen before starting the car solve my issue.

CryptoKylan, 2018-12-20:
yeah i dont really have an issue on starting the car although i do occasionally get an EPS fault upon start but almost never.  the problem is more random than that

i recorded my screen today showing what happens when i get the fault.  as you can see controlsd lags excessively.  any thoughts @rbiasini ?

it starts about 20 secs into the GIF

[image]

CryptoKylan, 2018-12-26:
I was able to downgrade NEOS back to the last version along with 0.5.6 and the EPS faults went away with my white panda.  HOWEVER with my grey panda and the antenna connected I still get EPS faults.  if i remove the antenna from the grey panda it works fine just like the white panda.  no faults at all.   

This is actually how my grey panda has acted since i got them in the summer, i just assumed it was an issue with me but now it seems like its more of a resource issue with the EON perhaps?  

with NEOS 8 on both my white and grey pandas I get EPS faults, antenna or not.  

the faults all happen the same way.  it appears to lag just enough to cause two lkasteeringcmd msgs to be delivered in under 0.01ms

heres proof it works fine with the grey pandas antenna disconnected.

one hour driving 
https://community.comma.ai/cabana/?route=7cc2a8365b4dd8a9%7C2018-12-25--15-00-48&max=55&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F7cc2a8365b4dd8a9%2F2b6fabb76e15a559425262c643d13387_2018-12-25--15-00-48

another hour driving back
https://community.comma.ai/cabana/?route=7cc2a8365b4dd8a9%7C2018-12-25--16-42-04&max=57&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F7cc2a8365b4dd8a9%2F25918245ddf8045d963383c69a414d3c_2018-12-25--16-42-04

heres the drive with the antenna connected where it didnt last even 15 mins which is about as long as i can get without a fault 
14:50:08 - 14:53:08 - 866 secs
https://community.comma.ai/cabana/?route=7cc2a8365b4dd8a9%7C2018-12-25--14-38-08&max=17&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F7cc2a8365b4dd8a9%2F87397ac015f523fded7dbb34783c0fd1_2018-12-25--14-38-08

im not sure what can be done about this.  any ideas?

CryptoKylan, 2019-01-03:
just an update.  last week I reverted back to neos 6 and both 0.5.6 and 0.5.7 work fine without any EPS faults.

couple hour drive without issue
https://community.comma.ai/cabana/?route=7cc2a8365b4dd8a9%7C2018-12-29--16-44-26&max=100&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F7cc2a8365b4dd8a9%2F136995c9b0879f75bccd5c53a4ad89b8_2018-12-29--16-44-26

this is still with my grey panda antenna disconnected.  if i plug the antenna in i will get an EPS fault after a few mins.  i even drove for an hour then plugged the antenna in while still driving and within a min or 2 my EPS faulted out.

as mentioned the faults all appear to happen the same way.  two lkasteeringcmd msgs in under 0.01ms as shown above

vtarasov mentioned profiling controlsd so i did.  heres the cprofile output file if it helps.

[acadia cprofile.zip](https://github.com/commaai/openpilot/files/2722402/acadia.cprofile.zip)

is there something you guys can suggest?

vanillagorillaa, 2019-03-10:
This should be resolved with 0.5.9 release. Going to close this but feel free to reopen if you still have issues.

closed 2019-03-10 by vanillagorillaa

Verylukyguy, 2022-01-25:
This issue has not yet been resolved.