commaai/openpilot #483: Drives do not start when PASSIVE="1"

braden337, 2018-12-25:
## Overview
I am using openpilot v0.5.7 (`release2` branch) with `PASSIVE="1"` on an Eon Gold in an unsupported vehicle with a Grey Panda.

## Problem
Drives no longer start when I turn on my vehicle, I noticed this after the NEOS v8 update. Prior to the update, when I turned on my vehicle, a drive would start.

## Troubleshooting
- If I set `PASSIVE="0"`, drives _do_ start when I turn on my vehicle.
- I uninstalled openpilot, and reinstalled on the Eon Gold with `https://openpilot.comma.ai`, and observed the same problem.
- I reset the local git repository to the openpilot v0.5.6 release commit [860a487](https://github.com/commaai/openpilot/commit/860a48765d1016ba226fb2c64aea35a45fe40e4a), and again observed the same problem.

## Resolution
The reason I am running openpilot with `PASSIVE="1"` is because my vehicle is unsupported and I wanted the latest "chffrplus" release. When you install chffrplus with `https://chffrplus.comma.ai` it pulls down version 0.4.7.2 from the `release` branch

Just now I  looked at the [chffrplus repo](https://github.com/commaai/chffrplus) and saw the `release2` branch was on 0.5.7 just like openpilot. So I thought I might as well try installing chffrplus with `https://chffrplus.comma.ai`, and then pull down version 0.5.7 manually.

I connected to the Eon over ssh and ran the following commands:

```shell
cd /data
rm -rf chffrplus
git clone https://github.com/commaai/chffrplus
reboot
```

I just went for a test drive, and when I started my vehicle, a drive started on the Eon and showed the UI for the camera calibration. 👍

vanillagorillaa, 2018-12-25:
Will debug this later this week. Pretty sure I know what's going on

closed 2018-12-26 by vanillagorillaa

braden337, 2021-12-17:
@vanillagorillaa I haven't used by Eon Gold in about a year, tonight I flashed NEOS and installed openpilot v0.8.12.

Set `PASSIVE=1` and rebooted.

I'm experiencing the same problem as stated [above](https://github.com/commaai/openpilot/issues/483#issue-393963746) when connected to my vehicle (drives don't start nor is the calibration ui shown to me)