commaai/openpilot #496: Lane keeping sway on left/right turn ramp

xuejunxie, 2019-01-09:
I have been driving with openpilot for 5K miles. The most dangerous situation I encounter is when the car approaches the left/right turn ramp. Openpilot seems to default to merge into the ramp if the ramp has a part without lane mark as in CA. It's fine if the ramp has dotted guide line as in WA. On local, the ramp leads to the opposite direction, which will lead the car into the wrong direction. On the highway, if the ramp has a long length without lane line, the openpilot will default turn into the ramp. If the ramp has a short length without lane line, the openpilot will sway in and out the ramp. I almost hit the barrier once.

Here is the image for the road in CA that lead to sway or going into wrong way:
[image]

Here is the ramp with a dotted guide in WA that works fine:
[image]

Another dangerous situation is lane keeping not centering in some situation. In most of the time, it will ride on the left side of the road if it's a left turn curve. In some time, my tire will touch the side of the lane even on a straight lane.

bugsy924, 2019-01-09:
Probably grab the wheel near unmarked exits 

either remount, recalibrate or manually add an angle offset to set your steer angle to 0 when in a straight line

pd0wm, 2019-05-27:
Exits are a known problem, and will be solved by future model updates.

closed 2019-05-27 by pd0wm