commaai/openpilot #498: Change CC speed via bluetooth

tritchie, 2019-01-11:
The CC stalk is an acceptable, but very inefficient, way to adjust OpenPilot’s target speed. 

It would be more convenient if the driver had the option to attach a small Bluetooth media button like this one somewhere (e.g. the console):
https://www.ebay.com/itm/Bluetooth-Media-Audio-Music-Remote-Control-Button-Car-Steering-Wheel-Bike-Mount-/391940366086
And use it to adjust the cruise control speed.

An example button scheme would be:
[>]=[increase cruise control by 5]
[<]=[cc -5]
[+]=[cc +1]
[-]=[cc -1]
[play/pause button] = [set cc to speed limit + delta, if applicable]

Bountysource link coming soon.

tritchie, 2019-01-11:
https://www.bountysource.com/issues/68575431-change-cc-speed-via-bluetooth

arne182, 2019-01-11:
Bluetooth is quite secure now a days. If you are scared then switch off the biggest vector the internet connection.

tritchie, 2019-01-15:
Also, there's a whole world of further opportunity if you incorporate long presses. For instance:
Press [>] = [cc+1mph]; Press [<]=[-1mph]
Hold [>;<] = [+5;-5], then +5/-5 every 1/2 second till button is released
Press [Play/pause button] = [set cc to 0]
Hold 2 seconds [play/pause] = [set to speed limit + offset]
Press [+] = [increase offset by 1]; press [-] = [decrease offset by 1]

ShaneSmiskol, 2019-02-01:
@tritchie How does this work? https://github.com/ShaneSmiskol/op-speed-controller-app

pd0wm, 2019-06-02:
Sounds like a nice gimmick, but not something we would upstream.

This issue has been inactive for a while, so I'm closing it.

closed 2019-06-02 by pd0wm