commaai/openpilot #506: Questions about the Toyota 2018 Corolla

Yifanny, 2019-01-18:
My car is Toyota 2018 Corolla with Electric Power Steering System and Auto Cruise Control System, and it is a hybrid car.

I have bought panda, giraffe, and EON. Since the model name of the Toyota in China is different from the USA, I do not know if there is TSS-P module in my car. In the introduction of my car, it does not indicate if it has TSS-P or not.

Could anyone inform me whether my car is supported by openpilot? 

Thanks.

lxalano, 2019-01-22:
Hi, I also want to consult, my car is a Chinese-made Corolla 2018, whether he is supported.

muyufan, 2019-02-03:
不支持，国内新一代卡罗拉还得等半年。

ucolchen, 2019-02-03:
问一下经销商，是否支持tss-p

hao-fu, 2019-03-14:
Is there any Corolla driver working on openpilot? I am open to communicate.

有没有卡罗拉车主交流交流怎么搞

bugsy924, 2019-03-14:
https://github.com/commaai/openpilot#supported-cars

legonigel, 2019-03-14:
I think this car might just require a new fingerprint. If so check out https://github.com/commaai/openpilot#how-can-i-add-support-for-my-car

ucolchen, 2019-03-14:
Hao-fu, please join the discord, there’s a lot of info and also a Chinese language sub group you might find people used same car

hao-fu, 2019-03-14:
@ucolchen Yes I joined discord. I guess you are "ucol", just sent you a friend request.

closed 2019-06-02 by pd0wm