commaai/openpilot #509: 2018 Honda Fit Support - User Port - Radar Issue

farisjarrah, 2019-01-25:
2019 Honda Fit EX models and higher have ACC and LKAS. I have bought the comma.ai hardware and started the port. However it seems as though I am having some issues with getting the radar to work.

I seem to be getting the error `videostream connect fail`

Here is a link to my work. I have the generated dbc file, copied from the 2018 honda fit one on energee's fit branch.

https://github.com/farisjarrah/openpilot/pull/1/files

Here is a link to a cabana drive, I had flashed openpilot so this is with the `visionstream connect fail` error

https://community.comma.ai/cabana/?route=fdf1ef5f7df528c6%7C2019-01-21--21-35-05&max=0&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2Ffdf1ef5f7df528c6%2Fa2b567c5bea924b20965f0af9f754d76_2019-01-21--21-35-05

I have tested energee's fit branch, however it did not seem to work. Also the fingerprints were different.

Any help looking where to troubleshoot or things to try in this matter would be greatly appreciated!

energee, 2019-01-25:
What is your switch setting? 1111?

energee, 2019-01-25:
Also confirmed fingerprints match, however previous fingerprint is using old format, so pushing an update (But that doesn't affect fingerprinting):

Diff based on attached drive:

```
diff --git a/selfdrive/car/honda/values.py b/selfdrive/car/honda/values.py
index 151266a..289dcab 100644
--- a/selfdrive/car/honda/values.py
+++ b/selfdrive/car/honda/values.py
@@ -102,7 +102,7 @@ FINGERPRINTS = {
     57: 3, 148: 8, 199: 4, 228: 5, 231: 5, 232: 7, 304: 8, 330: 8, 340: 8, 344: 8, 380: 8, 399: 7, 401: 8, 420: 8, 423: 2, 427: 3, 428: 8, 432: 7, 441: 5, 446: 3, 450: 8, 464: 8, 467: 2, 469: 3, 470: 2, 474: 8, 476: 7, 477: 8, 479: 8, 490: 8, 493: 5, 495: 8, 507: 1, 545: 6, 597: 8, 661: 4, 662: 4, 773: 7, 777: 8, 780: 8, 795: 8, 800: 8, 804: 8, 806: 8, 808: 8, 814: 4, 815: 8, 817: 4, 825: 4, 829: 5, 862: 8, 881: 8, 882: 4, 884: 8, 888: 8, 891: 8, 927: 8, 918: 7, 929: 8, 983: 8, 985: 3, 1024: 5, 1027: 5, 1029: 8, 1036: 8, 1039: 8, 1064: 7, 1108: 8, 1092: 1, 1115: 4, 1125: 8, 1127: 2, 1296: 8, 1302: 8, 1322: 5, 1361: 5, 1365: 5, 1424: 5, 1600: 5, 1601: 8, 1618: 5, 1633: 8, 1670: 5
   }],
   CAR.FIT: [{
-    57L: 3, 145L: 8, 228L: 5, 304L: 8, 342L: 6, 344L: 8, 380L: 8, 399L: 7, 401L: 8, 420L: 8, 422L: 8, 427L: 3, 428L: 8, 432L: 7, 464L: 8, 487L: 4, 490L: 8, 506L: 8, 597L: 8, 660L: 8, 661L: 4, 773L: 7, 777L: 8, 780L: 8, 800L: 8, 804L: 8, 808L: 8, 829L: 5, 862L: 8, 884L: 7, 892L: 8, 929L: 8, 985L: 3, 1024L: 5, 1027L: 5, 1029L: 8, 1036L: 8, 1039L: 8, 1108L: 8, 1322L: 5, 1361L: 5, 1365L: 5, 1424L: 5, 1600L: 5, 1601L: 8, 2015L: 3
+    57: 3, 145: 8, 228: 5, 304: 8, 342: 6, 344: 8, 380: 8, 399: 7, 401: 8, 420: 8, 422: 8, 427: 3, 428: 8, 432: 7, 464: 8, 487: 4, 490: 8, 506: 8, 597: 8, 660: 8, 661: 4, 773: 7, 777: 8, 780: 8, 800: 8, 804: 8, 808: 8, 829: 5, 862: 8, 884: 7, 892: 8, 929: 8, 985: 3, 1024: 5, 1027: 5, 1029: 8, 1036: 8, 1039: 8, 1108: 8, 1322: 5, 1361: 5, 1365: 5, 1424: 5, 1600: 5, 1601: 8
   }],
   # 2018 Odyssey w/ Added Comma Pedal Support (512L & 513L)```

farisjarrah, 2019-01-26:
For the giraffe(honda inverted nidec) I have tried the 3 main settings I could think of: 0000, 0111, and 1111. Surprisingly 1111 worked the best. I did not get LKAS errors or ACC errors. However on 0000 and 0111 I did get LKAS and ACC errors. On all of these settings I am not getting any speed readings and continued to get the `visionstream connect fail`

I just got done testing this new update from your branch @energee with the same 3 different giraffe settings with the same results. In addition, I did another port for the 2019 fit based off your branch with all the same settings in interface.py values.py and carstate.py except I changed the name to FIT_2019
I regenerated the fingerprint and regenerated the dbc file with these changes.

Both on your energee/fit branch as well as my own energee/fit/farisjarrah branches had the same results. 

What would be the proper way to go about troubleshooting this? I am happy to do the leg work and investigation, I just am a little stuck as to where to go next.

Tools I have: Workbench, smays, pandapaw, various cables to get it all hooked up. I Haven't gotten the smays to work under windows, or linux.

Really appreciate the help.

emmertex, 2019-01-27:
That is an error which has appeared since 0.5.8 for limited number of users.
Try going into visiond folder
/data/openpilot/selfdrive/visiond
and run
make clean
make

That may fix your issue

energee, 2019-01-28:
Once you confirm this issue is not related to visiond, please provide a drive using switch configuration: 0111 and i can identify any issues, if they still exist.

rbiasini, 2019-09-11:
Honda Fit support is merged in openpilot since 0.6.3. Please reopen if the issue persists on stock openpilot

closed 2019-09-11 by rbiasini