commaai/openpilot #510: is Visiond now fully open source ?

timotheeguerinrb, 2019-01-25:
In the last release, Version 0.5.8 (2019-01-17), we can see some code in the visiond parts and a comment on the release file that tell "Open source visiond".

So does it tell that now, openpilot is fully open source ?

I didn't see any announcement of that, but it's a great new if it's the case.

Thanks

dorkmo, 2019-01-25:
https://github.com/commaai/openpilot/tree/devel/selfdrive/visiond

emmertex, 2019-01-27:
visiond is, but the model isn't, infact, heaps isn't.
Openpilot is open source, but its worthless without a model.  
The model used to be inside visiond, now the model is by itself.

timotheeguerinrb, 2019-01-29:
I'm not looking for the model but I'm looking for porting the code to another hardware and try to make it less dependent of it.
What can be a problem when porting the code to a more generic hardware ?

kozuch, 2019-03-10:
I think all code (mainly C/C++) is now optimized for Snapdragon 820 platform. This means when porting to another platform you will have to deal with the differences. This means that porting within ARM platform will be easier than for example porting to x86. I have no experience in porting code between various ARM platforms, I am sure openpilot uses all possible HW acceleration like NEON (ARM's SIMD) etc. I have no idea how portable for exampe ARM's NEON code is.

closed 2019-06-17 by geohot