commaai/openpilot #521: 2019 Honda Pilot Fingerprint Issue

aleenoor1, 2019-02-04:
Hi, There is a fingerprint issue with all 2019 Honda Pilot. The car throws all kind of errors if a Comma Pedal is installed. Basically, the 512: 6, 513: 6 from the section CAR.PILOT_2019:[{ ~~~~

This goes in the file: /data/openpilot/selfdrive/car/honda/values.py

@wocsor can provide additional details but feel free to check with @aleenoor as well.

Thanks,

vanillagorillaa, 2019-02-21:
@aleenoor See above comments regarding adding fingerprint for Pilot

marcbou, 2019-04-24:
see https://github.com/commaai/openpilot/pull/613

with these changes the pedal works pretty well on my Passport 2019 which should be mostly identical to the Pilot 2019.

vanillagorillaa, 2019-04-24:
Fixed in next release, closing

closed 2019-04-24 by vanillagorillaa