commaai/openpilot #522: panda "make recover" shows link error "undefined reference to __aeabi_llsr"

kernyan, 2019-02-06:
In v0.5.8, panda's make recover shows following linker error

`arm-none-eabi-gcc -g -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -I inc -I ../ -nostdlib -fno-builtin -std=gnu11 -Os -Tstm32_flash.ld -o obj/bootstub.panda.elf obj/startup_stm32f413xx.o obj/bootstub.panda.o obj/sha.panda.o obj/rsa.panda.o
obj/sha.panda.o: 
In function SHA_final:
/.../openpilot/panda/crypto/sha.c:131: undefined reference to __aeabi_llsr
collect2: error: ld returned 1 exit status
`

This appears to be due to changes in panda/board/build.mk compile optimization flag change from -O2 to -Os. From what I gather, __aeabi_llsr is arm's long long shift right operation which was excluded by the compiler when the optimization flag changed.

adhintz, 2019-02-06:
Confirmed that this happens with `make clean && make recover` with -Os but does not happen with -O2

Apologies for breaking it.

kernyan, 2019-02-06:
@adhintz Yup that works! I actually arrived at the exact same fix 3dc9fb90f36ecfb99457c7aa5fd6cc1282a3964f but didn't know how to go about with it (my first issue created).

Thanks for fixing it too!

kernyan, 2019-02-08:
@adhintz @rbiasini 

While the linker flag did fix it when flashing from PC, it doesn't seem to work from EON. From EON, the compiler might complain about missing -lgcc. 

`arm-none-eabi-gcc -g -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -I inc -I ../ -nostdlib -fno-builtin -std=gnu11 -Os -Tstm32_flash.ld "-DALLOW_DEBUG" -o obj/bootstub.panda.elf obj/startup_stm32f413xx.o obj/bootstub.panda.o obj/sha.panda.o obj/rsa.panda.o -lgcc
/system/comma/usr/local/bin/../lib/gcc/arm-none-eabi/4.7.1/../../../../arm-none-eabi/bin/ld: cannot find -lgcc
collect2: error: ld returned 1 exit status
make: *** [build.mk:56: obj/bootstub.panda.bin] Error 1`

I found a solution to it, though the fix is rather ugly. See b59699cb1c8c793401dac948141968cc06c1cacc 

Thanks!

rbiasini, 2019-02-09:
The fix seems ok @kernyan. If it's verified that `make` and `make recover` work both on pc and EON I can merge the change.

kernyan, 2019-02-09:
With fix on sha.c

`make` on EON
[image]

`make recover` on EON
[image]
[image]

Without fix on sha.c
`make recover` on EON
[image]

Thanks!

rbiasini, 2019-02-09:
cool. I'll test independently and I'll merge. Thanks.

legonigel, 2019-04-09:
I believe this is fixed. Closing

closed 2019-04-09 by legonigel