commaai/openpilot #523: Issues with following distance

ShaneSmiskol, 2019-02-06:
This isn't an error, as it's working as programmed, but a mere suggestion. When someone cuts you off going 5 to 10 mph faster than you, op brakes the car to keep the following distance you set. op should instead take into consideration the relative speed of the car, so that it won't have to violently brake everytime someone overtakes you going considerably faster than you. This method of braking is also easy to see when you accelerate. Openpilot keeps a large following distance when accelerate from a stop, which it would be preferable and more natural to accelerate with the lead car, then keeping a distance when the car is at the set speed.

ShaneSmiskol, 2019-03-12:
I've implemented some of the stuff I talked about in this issue on my fork here: https://github.com/ShaneSmiskol/openpilot

If you could try it out and let me know how it performs, that would be great!

legonigel, 2019-03-13:
@ShaneSmiskol If you have fixes for actual issues, please open pull requests for the fixes. Each fix should be a separate PR. We welcome fixes from the community.

ShaneSmiskol, 2019-03-13:
@legonigel Got it! Once I fine tune it a little more I'll open a PR

closed 2019-04-27 by ShaneSmiskol