commaai/openpilot #541: dbc File overlapping

ashtonaureleo, 2019-02-26:
Hi,
We are trying to read CAN messages of Toyota Prius 2017 and translate into human readable using cantools package which uses the toyota_prius_2017_pt_generated.dbc file in opendbc folder.

Cantools.decode throws an error saying IPAS steering commands are overlapping.
Can the dbc file be edited to remove the error by rearranging the bits?
Will the ECU be able to read messages provided by cantools.encode based on the rearranged signals?

Could you suggest any other way to read CAN messages besides using cantools and dbc file?
Below is the Overlapping content of the Steering IPAS from dbc:

BO_ 359 STEERING_IPAS_COMMA: 8 IPAS
 SG_ STATE : 7|4@0+ (1,0) [0|15] "" XXX
 SG_ ANGLE : 3|12@0- (1.5,0) [-510|510] "deg" XXX
 SG_ SET_ME_X10 : 23|8@0+ (1,0) [0|255] "" XXX
 SG_ SET_ME_X00 : 31|8@0+ (1,0) [0|255] "" XXX
 SG_ DIRECTION_CMD : 38|2@0+ (1,0) [0|3] "" XXX
 SG_ SET_ME_X40 : 47|8@0+ (1,0) [0|255] "" XXX
 SG_ SET_ME_X00 : 55|8@0+ (1,0) [0|255] "" XXX
 SG_ CHECKSUM : 63|8@0+ (1,0) [0|255] "" XXX

legonigel, 2019-04-09:
Did you find any issue and fix with the DBC file? If so please open a PR in [commaai/opendbc](https://github.com/commaai/opendbc).

rbiasini, 2019-06-02:
I don't think those signals are overlapping. Are they? Remember, this is little endian.

ashtonaureleo, 2019-06-03:
The issue was due to SET_ME_X00 in steering IPAS id: 614message. There are two signals with the same name(SET_ME_X00). I have just renamed one of these signals as SET_ME_X00_edit.
[toyota_prius_2017_pt_generated_for_cantools.zip](https://github.com/commaai/openpilot/files/3246371/toyota_prius_2017_pt_generated_for_cantools.zip)

rbiasini, 2019-06-03:
ah yes, that's right. Let me fix it.

rbiasini, 2019-06-03:
fixed in https://github.com/commaai/opendbc/commit/34bd4c4dca459b02caba1b26eff7e2a703ebb423

closed 2019-06-03 by rbiasini