commaai/openpilot #555: Uncommanded disengagement

zeeexsixare, 2019-03-08:
On my drive in to work this morning, I was engaged and OP was functioning normally for about 10 seconds.  Ideal conditions. Then it suddenly disengaged and I couldn't tell except I started drifting out of the lane. And Green eon outline went away. Maps off. Path was dead straight. Speeds were around 40mph.  No sounds but I have engage/disengage sounds turned off. Cabana uploading, will share later.  This had happened before on a drive a few weeks ago but didn't occur again until now. 0.5.8

adhintz, 2019-03-08:
What car are you using? Do you have a cabana share link you can share?

zeeexsixare, 2019-03-10:
Hey Drew, the car is a 2017 Toyota Highlander ICE.  I've sent you a cabana link privately via Discord.

rbiasini, 2019-09-11:
@zeeexsixare normally we would dig into these issues (sorry we did not follow up months ago). However, the OP version you had the issue with is quite old now and it would not make much sense looking into it now. Closing the ticket.

closed 2019-09-11 by rbiasini