commaai/openpilot #556: 2018 Volt not working as of v0.5.9

srpape, 2019-03-08:
If I revert to the commit just BEFORE v0.5.9 (fdb04d9f69cf7d088149d76b7433e59501dfe688), everything works as expected. v0.5.9 (0207a970400ee28d3e366f2e8f5c551281accf02), or after does not work.

When not working, OP detects that the car is running, but the cruise buttons do nothing. The service front camera/ACC messages also appear when the car is started.

srpape, 2019-03-08:
Watching from tmux, I've seen both:
```
fingerprinted CHRYSLER PACIFICA 2018
```

as well as 

```
car doesn't match any fingerprints: {528L: 4, 532L: 6, 546L: 7, 550L: 8, 554L: 3, 558L: 8, 560L: 8, 562L: 8, 563L: 5, 564L: 5, 565L: 5, 566L: 5, 567L: 3, 568L: 1, 573L: 1, 1105L: 6, 1618L: 8, 608L: 8, 609L: 6, 610L: 6, 611L: 6, 612L: 8,
613L: 8, 1265L: 8, 647L: 3, 1221L: 5, 1187L: 4, 170L: 8, 171L: 8, 1225L: 7, 189L: 7, 190L: 6, 193L: 8, 707L: 8, 197L: 8, 199L: 4, 201L: 8, 1217L: 8, 1227L: 4, 209L: 7, 211L: 2, 1249L: 8, 1233L: 8, 1257L: 6, 241L: 6, 1267L: 1, 761L: 7, 127
5L: 3, 298L: 8, 1280L: 4, 1300L: 8, 1322L: 6, 288L: 5, 1328L: 4, 810L: 8, 1323L: 4, 304L: 1, 309L: 8, 311L: 8, 313L: 8, 320L: 3, 840L: 5, 842L: 5, 844L: 8, 352L: 5, 866L: 4, 711L: 6, 1905L: 7, 1906L: 7, 1907L: 7, 1910L: 7, 489L: 8, 1912L:
 7, 381L: 6, 1223L: 3, 386L: 8, 388L: 8, 389L: 2, 390L: 7, 1927L: 7, 1417L: 8, 1005L: 6, 508L: 8, 417L: 7, 419L: 1, 1009L: 8, 426L: 7, 328L: 1, 961L: 8, 451L: 8, 452L: 8, 453L: 6, 454L: 8, 967L: 4, 456L: 8, 969L: 8, 977L: 8, 979L: 7, 1273
L: 3, 988L: 6, 989L: 8, 479L: 3, 2016L: 8, 481L: 7, 995L: 7, 2020L: 8, 485L: 8, 2024L: 8, 1001L: 8, 2028L: 8, 493L: 8, 495L: 4, 497L: 8, 499L: 3, 500L: 6, 501L: 8, 1017L: 8, 1019L: 2, 1020L: 8}
```

rbiasini, 2019-03-08:
The fingerprint for your car is probably incomplete. Could you try to collect the right fingerprint?
https://community.comma.ai/wiki/index.php/Creating_Fingerprints

adhintz, 2019-03-08:
We might also be able to remove this particular Pacifica fingerprint: https://github.com/commaai/openpilot/blob/devel/selfdrive/car/chrysler/values.py#L26

That Pacifica fingerprint was generated by someone running an OBD2 double-bypass system connected directly to the CAN distribution box behind the glovebox. Something like this: https://zautomotive.com/product/double-bypass/
That person now has a more typical Comma FCA Giraffe. Both @srpape 's fingerprint and the double-bypass Pacifica fingerprint have a suspiciously similar set of messages that I haven't seen before.
In addition to @srpape adding a fingerprint for their car, I'll ask the person with a Pacifica double-bypass if we can remove the `{170: 8, 171: 8, 189: 7,` Pacifica fingerprint.

srpape, 2019-03-08:
I've been using the car since September without issue until that one particular commit.

Adding to the fingerprint did seem to help with v0.5.9 somewhat, but fingerprinting doesn't seem to finish until after the car has already issued the warnings now.

Needs more testing, but when I updated to the latest code I wasn't seeing fingerprinting even finish. Just "waiting for fingerprint...". The logs are kind of a pain, is there a better way to get them than watching tmux?

```
diff --git a/selfdrive/car/gm/values.py b/selfdrive/car/gm/values.py
index 27ca73ff..12d6b57d 100644
--- a/selfdrive/car/gm/values.py
+++ b/selfdrive/car/gm/values.py
@@ -72,7 +72,7 @@ FINGERPRINTS = {
   },
   # Volt Premier w/ ACC 2018
   {
-    170: 8, 171: 8, 189: 7, 190: 6, 193: 8, 197: 8, 199: 4, 201: 8, 209: 7, 211: 2, 241: 6, 288: 5, 298: 8, 304: 1, 308: 4, 309: 8, 311: 8, 313: 8, 320: 3, 328: 1, 352: 5, 381: 6, 384: 4, 386: 8, 388: 8, 389: 2, 390: 7, 417: 7, 419: 1, 426: 7, 451: 8, 452: 8, 453: 6, 454: 8, 456: 8, 479: 3, 481: 7, 485: 8, 489: 8, 493: 8, 495: 4, 497: 8, 499: 3, 500: 6, 501: 8, 508: 8, 528: 4, 532: 6, 546: 7, 550: 8, 554: 3, 558: 8, 560: 8, 562: 8, 563: 5, 564: 5, 565: 5, 566: 5, 567: 3, 568: 1, 573: 1, 577: 8, 608: 8, 609: 6, 610: 6, 611: 6, 612: 8, 613: 8, 647: 3, 707: 8, 711: 6, 715: 8, 717: 5, 761: 7, 810: 8, 840: 5, 842: 5, 844: 8, 866: 4, 869: 4, 880: 6, 961: 8, 967: 4, 969: 8, 977: 8, 979: 7, 988: 6, 989: 8, 995: 7, 1001: 8, 1005: 6, 1009: 8, 1017: 8, 1019: 2, 1020: 8, 1033: 7, 1034: 7, 1105: 6, 1187: 4, 1217: 8, 1221: 5, 1223: 3, 1225: 7, 1227: 4, 1233: 8, 1249: 8, 1257: 6, 1265: 8, 1267: 1, 1273: 3, 1275: 3, 1280: 4, 1296: 4, 1300: 8, 1322: 6, 1323: 4, 1328: 4, 1417: 8, 1601: 8, 1905: 7, 1906: 7, 1907: 7, 1910: 7, 1912: 7, 1922: 7, 1927: 7, 1930: 7, 2016: 8, 2020: 8, 2024: 8, 2028: 8
+    170: 8, 171: 8, 189: 7, 190: 6, 193: 8, 197: 8, 199: 4, 201: 8, 209: 7, 211: 2, 241: 6, 288: 5, 298: 8, 304: 1, 308: 4, 309: 8, 311: 8, 313: 8, 320: 3, 328: 1, 352: 5, 381: 6, 384: 4, 386: 8, 388: 8, 389: 2, 390: 7, 417: 7, 419: 1, 426: 7, 451: 8, 452: 8, 453: 6, 454: 8, 456: 8, 479: 3, 481: 7, 485: 8, 489: 8, 493: 8, 495: 4, 497: 8, 499: 3, 500: 6, 501: 8, 508: 8, 528: 4, 532: 6, 546: 7, 550: 8, 554: 3, 558: 8, 560: 8, 562: 8, 563: 5, 564: 5, 565: 5, 566: 5, 567: 3, 568: 1, 573: 1, 577: 8, 608: 8, 609: 6, 610: 6, 611: 6, 612: 8, 613: 8, 647: 3, 707: 8, 711: 6, 715: 8, 717: 5, 761: 7, 810: 8, 840: 5, 842: 5, 844: 8, 866: 4, 869: 4, 880: 6, 961: 8, 967: 4, 969: 8, 977: 8, 979: 7, 988: 6, 989: 8, 995: 7, 1001: 8, 1005: 6, 1009: 8, 1017: 8, 1019: 2, 1020: 8, 1033: 7, 1034: 7, 1105: 6, 1187: 4, 1217: 8, 1221: 5, 1223: 3, 1225: 7, 1227: 4, 1233: 8, 1249: 8, 1257: 6, 1265: 8, 1267: 1, 1273: 3, 1275: 3, 1280: 4, 1296: 4, 1300: 8, 1322: 6, 1323: 4, 1328: 4, 1417: 8, 1601: 8, 1618: 8, 1905: 7, 1906: 7, 1907: 7, 1910: 7, 1912: 7, 1922: 7, 1927: 7, 1930: 7, 2016: 8, 2018: 8, 2020: 8, 2024: 8, 2028: 8
   }],
   CAR.BUICK_REGAL : [
   # Regal TourX Essence w/ ACC 2018
```

srpape, 2019-03-09:
Above MR seems to resolve the issue for me, as long as it's okay to remove the Pacifica fingerprint as per @adhintz

PR #561 (2019-03-10): Fix 2018 volt fingerprint
Update fingerprint and remove offending PACIFICA_2018 match.

This resolves the issue for me, but we have to confirm that it's okay to remove the Pacifica fingerprint.

Fixes #556
closed 2019-03-10 by rbiasini