commaai/openpilot #562: Porting to another hardware

kozuch, 2019-03-10:
The visiond was open sourced lately. Does anyone have an idea what port to another hardware could be valuable somehow? The simplest could be another Snapdragon 820 phone which may be cheaper than EON Dev kit. The full porting would require also a eon-neos port I guess.

rjschof, 2019-03-10:
Where did you see that visiond was open sourced recently? Can you post a link to the repo?

kozuch, 2019-03-10:
It has been announced on twitter: https://twitter.com/comma_ai/status/1089041100285726721

The visiond folder now contains code: https://github.com/commaai/openpilot/tree/devel/selfdrive/visiond

As mentioned elsewhere (#510) the only binary (=proprietary) part of openpilot now shoud be the DNN models (see /models) - these are in .dlc format which I have no knowledge about (but I guess shoudl be portable since the twitter post hopes for ports now).

NeoFromMatrix, 2019-03-10:
Three options come in mind:

* x86_64 and just take any small pc you can find. Could be hardware agnostic. If the cross compatibility can be solved this might be broaden the community of developers. 
* some arm dev board (will depend on SW distro and RT patch availability/ease of integration)
* another Android phone (requires RT patch)

In reality I guess it depends on the lead developers and whatever they come up with or

legonigel, 2019-03-11:
Openpilot currently only supports the 820. I would recommend that discussions regarding porting or different hardware continue on [discord](discord.comma.ai). Join the discord and check out the channels #hardware-unofficial and #topic-open-visiond

closed 2019-03-11 by legonigel