commaai/openpilot #568: Can openpilot determine turn left or right in next cross?

jinfagang, 2019-03-15:
Just wonder, does it knows where I (my car) am? Will it continue determine which cross to turn ? How did it knows where I will go?

bugsy924, 2019-03-15:
No

jinfagang, 2019-03-15:
@bugsy924 Then it just L2

jfrux, 2019-03-15:
Correct, it’s always been L2 and will be for quite some time.
There aren’t any L3 cars available for purchase.
The good news is, that the software can reach L3 long before your car will ever be L3.
So if you switch cars, you can port it and upgrade any supported car with this at any time.

jinfagang, 2019-03-15:
@jfrux I wanna build my own self driving car (whole car tiny not just a plugger) with L3 capable

jfrux, 2019-03-15:
You and the rest of the world mate!
Best of luck to you!

dorkmo, 2019-03-15:
@jinfagang join the discord, there is a channel discussing traffic light and stop sign detection. Possiblly L2.5

icmma, 2019-03-30:
@jinfagang  by end of the year OP will be L3:)

jinfagang, 2019-03-30:
@dorkmo  I am not mean that, without navigation and location it can not know where to go

ucolchen, 2019-03-30:
I think that's part of the big picture, i see right now peoples are working hard on the puzzle piece by piece, and achieved great milestones, lkas, acc, stop and go, auto lane change, hd maps...great job comma community doing here!

jinfagang, 2019-03-30:
BTW, does there any deeplearning models summary used in comma? such as object detection model, lane   and drivable area segmentation model, and lidar & camera fusion model etc.

legonigel, 2019-03-31:
No. If you want to see what models currently are used, you can check out the code in `selfdrive/visiond`

closed 2019-06-02 by pd0wm