commaai/openpilot #574: Add support "2019 Honda CRV Hybrid" work?

gzjackar, 2019-03-21:
Background info:
1、My car is 2019 Honda CRV Hybrid which is not supported by Comma yet.
2、I get my car fingerprint and replace the CRV_5G fingerprint in values.py files, At first, when I engage OP ,I got the “openpilot unavailable ,Door Open”  ,so I set the ”self.door_all_close=True” in carstare.py . Secondly, I got the “Openpilot unavaiable, Gear not D” error ,
Now I don’t know what should I do now
3、I got the dump.py carState file, could someone help to analyze it?
4、How can I make comma support the new Model “2019 Honda CRV Hybrid” ?
5、I have upload the drive log and my car dbc file into cabana link ,does it help ?what should I do in the next step?

https://community.comma.ai/cabana/?route=232585b7784c1af4%7C2019-03-15--18-21-58&max=19&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F232585b7784c1af4%2F939a87aac9a2bb080196dce848533321_2019-03-15--18-21-58

https://github.com/gzjackar/opendbc/commit/e0b13a51518690da5cd2db7f0d5b84e572478bba

gzjackar, 2019-03-21:
"2019 Honda CRV Hybrid “ fingerprint
fingerprint 57: 3, 148: 8, 228: 5, 304: 8, 330: 8, 344: 8, 380: 8, 387: 8, 388: 8, 399: 7, 419: 8, 420: 8, 427: 3, 432: 7, 441: 5, 450: 8, 464: 8, 477: 8, 479: 8, 490: 8, 495: 8, 525: 8, 531: 8, 545: 6, 662: 4, 773: 7, 777: 8, 780: 8, 804: 8, 806: 8, 808: 8, 814:
4, 829: 5, 833: 6, 862: 8, 884: 8, 891: 8, 927: 8, 929: 8, 1302: 8, 1361: 5, 1365: 5, 1600: 5, 1601: 8

vanillagorillaa, 2019-03-21:
You may want to work with some of the community members on our discord, discord.comma.ai. You will want to find the door message and the gear packet for this car. Once you have those you can add them in the DBC and update the corresponding openpilot files. Once you have those you can submit the pull requests to add support for your CR-V. There are individuals on discord that would probably be very interested in helping you get this setup.

energee, 2019-03-22:
Can you post a cabana drive with EON detached and recording you opening door and close. This signal looks ok to me above.

energee, 2019-03-22:
https://github.com/commaai/opendbc/pull/148

gzjackar, 2019-03-23:
> Can you post a cabana drive with EON detached and recording you opening door and close?

> You may want to work with some of the community members on our discord, discord.comma.ai. You will want to find the door message and the gear packet for this car. Once you have those you can add [...]

thank u~

gzjackar, 2019-03-23:
> Can you post a cabana drive with EON detached and recording you opening door and close?

I want to share the cabana video with recording the door opening and closing , but the eon screen shows the black screen  , how can I do this?
[image]

gzjackar, 2019-03-23:
> Can you post a cabana drive with EON detached and recording you opening door and close?

I use the can printe to try to tell which number affect door-open-close , but all the numbers are keep changing all the time , so I can't tell which number is?
[image]

gzjackar, 2019-03-24:
> Can you post a cabana drive with EON detached and recording you opening door and close?

Hi energee, I have found the 0:374 number 00(door closed) a1(door open) change ,and next?
[image]

https://community.comma.ai/cabana/?route=232585b7784c1af4%7C2019-03-24--10-52-02&max=28&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F232585b7784c1af4%2Fe9536202d6d275797dc3cc96dcc1dde1_2019-03-24--10-52-02
[image]

energee, 2019-03-25:
Nice find, want to try opening other doors? I see two bits that flip so one might bee door ajar and one might be driver door or similar?

gzjackar, 2019-03-28:
> Nice find, want to try opening other doors? I see two bits that flip so one might bee door ajar and one might be driver door or similar?

Other doors will also change the number from 00 to a1

energee, 2019-03-29:
:/ sounds like new can signals, anyway we can work with that

gzjackar, 2019-03-30:
> :/ sounds like new can signals, anyway we can work with that

Yes, CRV Hybrid is only published in China, Japan , England, So it's not supported officially yet.
Now efini had help me to run with it in the incomplete DBC mode. 
https://community.comma.ai/cabana/?route=232585b7784c1af4%7C2019-03-29--20-07-08&max=19&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F232585b7784c1af4%2F507f04b1e2b100d1e1e9c9a83eb2de7e_2019-03-29--20-07-08

gzjackar, 2019-03-30:
> Yes, CRV Hybrid is only published in China, Japan , England, So it's not supported officially yet.
> Now efini had help me to run with it in the incomplete DBC mode. [...]

And with the part of the CRV 5G DBC file , the car's steer move not very smoothly in the low speed.  I think the other part works like the normal way.

eFiniLan, 2019-04-18:
The port has been merged, safe to close this now :)

gzjackar, 2019-04-18:
> The port has been merged, safe to close this now :)

OK,Great thanks to you ^-^

closed 2019-04-18 by gzjackar