commaai/openpilot #576: Hummer H2

supergenio, 2019-03-24:
Hi I’m pretty new at this but does anyone know if there’s a way to make it work on a Hummer H2 2003

energee, 2019-03-25:
My guess that might be a stretch as 2003 is pretty old, does H2 have adaptive cruise control and lane keeping assist?

legonigel, 2019-03-29:
I think you might want to check out the [discord](discord.comma.ai). You can find channels there to discuss with the community. You might want to checkout the channel `#topic-without-lkas`

ucolchen, 2019-03-29:
> My guess, that might be a stretch as 2003 is pretty old, does H2 have adaptive cruise control and lane keeping assist?

learned from discord, #topic-without-lkas, as soon as the car has EPS, there's chance op will support in future.

icmma, 2019-03-30:
it will be :) SoonTM @ucolchen

closed 2019-04-25 by legonigel