commaai/openpilot #578: Openpilot not disengaging with radar issue (GM volt)

srvolt, 2019-03-25:
I have a 2017 Volt, and in the middle of a drive OP stopped detecting the lead car, but didn't disengage.  I manually disengaged when it became clear that there was an issue.  I was able to operate OP normally, set speed limits etc, but OP would just behave as if there was no lead car.  

Apparently this is related to the radar in the Volt needing to be recalibrated, the fix is to remove the Giraffe and put the ASCM back in and drive around for a bit (in my case 200+ miles) and wait for the stock adaptive cruise control to come back online.

The main issue is that OP didn't disengage or provide any notification that something was wrong.  Stock ACC doesn't allow ACC to be engaged, and provides a notice that says 'ACC temporarily unavailable).  Seems like this could be a safety issue that I wanted to point out, let me know if you need any additional information. Thanks,

Raul

legonigel, 2019-03-25:
Would you be able to provide or PM me on on discord a link to explorer or cabana for the drive where this happened?

vntarasov, 2019-04-07:
@srvolt Can you please add the error bit that you had, to the existing GM radar health check at https://github.com/commaai/openpilot/blob/devel/selfdrive/car/gm/radar_interface.py#L76 ?

srvolt, 2019-04-14:
> @srvolt Can you please add the error bit that you had, to the existing GM radar health check at https://github.com/commaai/openpilot/blob/devel/selfdrive/car/gm/radar_interface.py#L76 ?

@vntarasov Sorry - I'm not sure where to go to find that information, if you don't mind pointing me in the right direction I'm more than happy to help.  Thanks

rbiasini, 2019-09-11:
@srpape @vntarasov , were you guys able to find the error bit?

vntarasov, 2019-09-11:
@rbiasini  I think saw "misaligned fault" bits in one of the logs people sent me in the past, but don't remember 100% which bit, and I don't have the link anymore.
I think it was FLRRMsalgnYawRt / FLRRMsalgnYawLt / FLRRMsalgnPtchUp / FLRRMsalgnPtchDn on the F_LRR_Obj_Header message. Currently `car/gm/radar_interface.py` doesn't check for those.

adeebshihadeh, 2021-06-02:
Closing this since radar will only be used offline (see #21070), though we'd merge a PR to that handles the fault.

closed 2021-06-02 by adeebshihadeh