commaai/openpilot #586: Middle like missing

arne182, 2019-03-29:
If the middle line is unavailable then the car should hug right.
Especially if the lane width now measures double the standard at that speed.

BatmanTSLA, 2019-05-27:
Vote on that, common „issue“ in germany. Narrow streets, no center line. Car is tends to move to the middle of the road.

Martint1980, 2019-07-13:
please add it to the new version

ErichMoraga, 2019-07-13:
Ugh, just when we're ranting about all the hugging issues, you want MORE hugging. :sad:

arne182, 2019-07-13:
I have an idea for a fix for this issue. Something like, if OSM has 2 lanes for this street and vision measures a value that is obviously double  that of a narrow road width, it ignores the left lane line. 

Or just get a bucket of paint and paint the middle lane line.

ErichMoraga, 2019-07-13:
Bucket of paint has been on my "bucket list" of things to do...

pd0wm, 2019-08-16:
0.6.3 has improved behavior for lines wider than a normal lane

closed 2019-08-16 by pd0wm