commaai/openpilot #594: Support Lexus RX300 2018?

but0n, 2019-04-07:
Hi,
I've bought a EON DevKit, panda and giraffe. I wonder is this tools support on the Lexus RX300 2018? I didn't seen this car on the list.

ucolchen, 2019-04-07:
have you tried on the car? discord is a better place to discuss and for more information. 
https://opc.ai/discord

but0n, 2019-04-09:
@ucolchen 
No, I haven't. I asked on discord, seems like nobody tried on the car.

eFiniLan, 2019-04-09:
@but0n it will probably not work out of box but since OP already support RX450h, it won't be too hard to make it work on your car.

but0n, 2019-04-09:
@eFiniLan Thanks! I want to join the wechat group, can you share me the QR code?

eFiniLan, 2019-04-16:
This can be closed, RX300 (China version)  can use rx450h port with minor modification (checks[] in carstate.py)

closed 2019-04-16 by but0n