commaai/openpilot #598: Uploader sort order issue

Gernby, 2019-04-09:
It seems that there is a defect in the directory sorting within uploader.py that results in older directories and files not being cleaned up.  I suspect it has something to do with the listdir_by_creation_date() not actually sorting by creation date, but by last modified date instead.  Basically, if a number of days worth of drives backs up on the EON, and upload speed of the internet connection isn't fast enough to actually catch up, things go haywire.  I have files and folders that are over a week old, yet I see drives that have uploaded from drives today in cabana.  I've also noticed that the timestamps on the directories don't match the folder names AT ALL.
[image]

pd0wm, 2019-06-02:
I assume the modified date was changed when logs were bzipped. Currently this is done on the fly.

Therefore I assume this issue is now fixed. Feel free to reopen when you still have problems.

closed 2019-06-02 by pd0wm

Gernby, 2019-06-03:
I don't think the zipping is the cause of the issue.  It seems that the uploader is uploading the rlog files from all existing directories, then uploading the video files from the directories.  It deletes the rlog files as it goes, which updates the modified date of the directory.  If the upload speed is not fast enough to completely empty the sdcard folder before the next drive occurs, the existing video files will remain until after the rlog files from the next drive get uploaded.  In my case, I was accruing days worth of video files that would never upload, so I just had to keep purging them manually.

I think the fix for this would be to A) fix the directory sort issue, and B) upload the rlog and video files as pairs.