commaai/openpilot #599: Account for Variable Rate Steering on Prius Prime to fix oscillations/twitchy lateral control

zorrobyte, 2019-04-11:
> Give the age of this issue and recent changes in controls. If the issue still affects the latest release, it can be reopened or a new issue opened. https://github.com/commaai/openpilot/issues/300

I'm a new OP user with a '19 Prius Prime. Running stock latest stable OP, autotune variables settle at:
/data/params/d {"steerRatio": 16.329651149572307, "carFingerprint": "TOYOTA PRIUS 2017", "stiffnessFactor": 0.5, "angleOffsetAverage": 1.0404464786796497} {"angle_model_bias": 0.3890739310530656}

End to end ratios per factory specs is ```The 2017-2018 Prius Prime New Car Features book (available by subscription to techinfo.toyota.com) says the steering ratio for all grades of the Prius Prime sold in the U.S. is 13.4.```

Prius Prime steering is rather twitchy and while I don't have experience with OP on other vehicles, @zeeexsixare 's tuning pdf makes me believe that this behavior is undesirable https://github.com/commaai/openpilot/pull/464

I am very aware of the work @Gernby @ErichMoraga are putting into lateral control, with Erich's Variable Rate Steering branch specifically addressing Prius Prime's lateral control; however, I believe an issue should stay open and updated on the main repo as a longterm upstream fix will be required at some point.

zorrobyte, 2019-04-16:
Another '19 Prius Prime extreme see-saw though that same turn as previously reported. 

Clean data, good weather, ran 0510 stock 17:33:13 - 17:36:13 and 253 -- two turns 5e7462b4437e3372|2019-04-16--17-30-13&seekTime=193 

https://community.comma.ai/cabana/?route=5e7462b4437e3372%7C2019-04-16--17-30-13&max=7&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F5e7462b4437e3372%2F7e75afc578e2867b4be8d0b96414b655_2019-04-16--17-30-13

I can replicate that test loop anytime, let me know if it's needed or want to collab.

ErichMoraga, 2019-04-16:
Stepping away from stock code, let me know in Discord how you like my latest VSR port.

mtourangeau, 2019-04-16:
I’ll try it.

zorrobyte, 2019-04-17:
@energee @zorrobyte This sounds like its related to the stiction issue with prius, are you already aware of this?

zorrobyte
@energee could you elaborate by what you mean by stiction?

Additional feedback: tried Grenbys feed forward fork last night and had a buttery smooth straits and turning experience on the same test loop, that curve which has always been choppy. It was the first time I’ve ever felt “on rails”, giving me hope this issue can be resolved within upstream OP

csouers
Static friction. Need to apply so much torque to break free and allow movement.

zorrobyte
Hm, I can't say I've really had much experience or have heard much about Prius's actuators. What I can say is that OP on Prius Prime constantly adjusts steering and is very overactive. It's almost like the planner doesn't plan lateral control ahead and instead respond to every single little change the vision system picks up.

Another interesting point is that the curve above that I use to test; OP can clearly see the lines but just seesaws desired angle seemingly for no reason, sending the car bouncing all over the place.

[image]

Desired angle should just hold steady, vision sees the curvature of the road, it's constant and predictable.

Even with model specific implementation issues, I'd imagine desired angle is "upstream" of vehicle specific lateral control? I could be wrong.

energee, 2019-04-17:
IRC, Comma has a Prius which should have the same EPS as Prius Prime, @ErichMoraga can probably confirm.

zorrobyte, 2019-04-17:
Some additional community chatter:
```
@Gernby  After the discovery last weekend that the PID / PIF values from latcontrol.py are almost never sent to the Toyota's EPS without being clipped by downstream logic, it's clear that the steering issue really isn't caused by laggy or sticky actuators.  We just need to improve the tuning of the PID values, and provide feedback into latcontrol when the requested torque is being clipped.  Otherwise, the actuators are just going to be controlled entirely by riding the maximum extents of a validation, which sucks.
The reason why Flag, VSR, Resonant, and the rate-based feedforward improves straight line driving so much for the Prius is because they allow the torque requests to be metered well enough that it stays within the validation boundaries.

foglem
Sounds hopeful!

Gernby
Actually, I shouldn't use the term "clipped" to describe what the Toyota validation logic is doing, since it doesn't just reduce the requested torque value.  Sometimes it actually increases the requested torque, since there are limits to the rate of change in both directions from sample to sample.```

zeeexsixare, 2019-04-17:
The stock Highlander tuning uses RAV4 parameters and is very choppy even on the straights at low speed <30mph.  Turns are just as bad.  Increased speed helped, but the final fix was to edit the PI parameters.  I suggest using my fork's Highlander params from #464 just to see if the significantly different params give any improvement, or worsen it.

By the way, I strongly feel that the stiction/lag problem is primarily lag.  You can compare timestamps of CAN data showing torque CMD and actual steering angle.  The other reason I feel that it is NOT stiction is that the stock LDA controller can implement very finely controlled steering with very low torque such as when the LDA triggers, or when the stock camera detects a curve ahead and assists in anticipation.

legonigel, 2019-04-25:
The variable steer ratio should not cause any issues or oscillations. Almost all modern cars have this, and this is why the steer ratio learner was added in 0.5.10. Variable steer ratios are only a problem at large steer angles. 

We did try our prius and found we have a similar twtichiness. We are trying to develop a solution, but it might take some time.

zorrobyte, 2019-04-26:
There's a sensor delay. It isn't the actuator.

[image]

[image]

5e7462b4437e3372|2019-04-24--17-26-01&seekTime=294

ErichMoraga, 2019-04-26:
Let's replace the sensor!

zorrobyte, 2019-04-26:
[image]

Hold my beer

zorrobyte, 2019-04-28:
Root cause analysis:
There's a rubber damper between the EPS motor and steering shaft. It has 0.4mm of backlash or more.

I 3D printed a solid replacement. Steering response is excellent now.

Before:
[image]
This one is really bad:
[image]

After:
[image]

Video of the "Dampner":
https://youtu.be/jloNX67sLR8

Video of my unscientific testing:
https://youtu.be/njIcvTiDFYI

zorrobyte, 2019-04-28:
[image]

zeeexsixare, 2019-04-30:
This is incredible work!  It's been dogging us for forever and I don't think anyone here tore down the assembly to find the root cause.  Looks like you have some photos; any way we can get a full step-by-step teardown set of photos and work instruction?  Would be great to see if it's minimally invasive like disconnecting DSU or if it's brutal like an engine teardown.

ErichMoraga, 2019-04-30:
Have you see the YouTube video?  This is the same cogwheel I showed everyone 2 years ago in Slack, but was not going to tear apart my car when it was new.

ErichMoraga, 2019-04-30:
Here's his video on this, bearing in mind only select Toyotas have this cogwheel damper...
https://youtu.be/jloNX67sLR8

zeeexsixare, 2019-04-30:
I haven't, no. Linky?  Your car is old now, OK to tear apart :)

ErichMoraga, 2019-04-30:
The YouTube link is a few posts up, and yeah, I'll be posting a step-by-step on the mod, but I don't believe Highlanders have this cogwheel "damper".

Gernby, 2019-05-01:
I agree that the OEM bushing in the steering column probably contributes to poor OP steering performance.  However, the only "delay" that it could cause would be in the "free play" region.  Once the steering rotates enough to close the gap, the remaining delay associated with the bushing would be eliminated.

zorrobyte, 2019-05-01:
Yes, until reverse torque is applied which happens frequently with the PI control scheme, which is to be expected in this application. The math of the small amount of time the rather high inertia EPS motor is freewheeling, the delay from a forward/reverse torque, and the sensitivity/reaction time of the PI controller is beyond me, but noticeably affects practical performance.

There's also backlash/elasticity in the steering shaft, thanks to two U-joints which is unfortunate.

I wish I was smarter and could approach this scientifically. Alas, I'm just some idiot hacking his CAN bus and installing 3D printed parts in a car with less than 2,000 miles 😆

Gernby, 2019-05-01:
@zorrobyte This is great work, so I hope my comment didn't come across as a criticism.  I just wanted to point out that there are multiple "stacked" factors contributing to this issue.  There's not going to be a magic bullet, and software workarounds are almost certainly going to require trade-offs in other performance categories.

Gernby, 2019-05-01:
@zorrobyte The free play in the steering column obviously causes a dead-zone in the reported steering angle.  That seems like a good opportunity for a dead-zone adjustment in the logic.  I recall you saying that you observed benefit from a deadzone into PID, but I assume that the deadzone value you used was a positive number.  If so, that would tend to cause under steer AND oversteer.

It seems that the case of a **mechanical** deadzone might require a **negative logical** deadzone.  Basically, if there is 1.0 degree of free play in the sensor data, then -0.5 degrees of logical deadzone might be beneficial.  However, without having a Toyota to test with, I can only guess.  If that doesn't turn out to be beneficial, then my next guess would be that a negative deadzone should be applied only when desired angle is closer to center than actual.

zorrobyte, 2019-05-05:
@pd0wm Partially, if not totally solved with replacing the EPS Damper & tuning params based on steeringAngle https://youtu.be/rHp__gzwzlY Great results on Rav 4 without hardware modifications: https://discordapp.com/channels/469524606043160576/534769493734588447/574467722100998165
0.4+/- degree oscillations in corners, +/- 0.5mm in straights.
https://github.com/zorrobyte/openpilot -- devel_modStockPID branch. My code changes are arguably minimal.

Compare this to previous screenshots of Grafana data ^^
[image]
[image]

Note: You MUST revert the steeringRatio in your `data/params/d/LiveParameters` to around 14 for Prius (13.40 preferable) 18.25 for Rav 4. I based my tunings at low SR, most stock OP params settle at 20-22, drastically overscaling my PID values.

I also added a derivative term to your PI control. I only use it for extreme angles at torque limits to smooth out OP, that and Proportional on Measurement (self.proportional_on_measurement = False # http://brettbeauregard.com/blog/2017/06/introducing-proportional-on-measurement/) could be worth looking into for further experimentation.

Cheers!

zorrobyte, 2019-05-09:
Due to news of Comma switching out PID for INDI Control (which could have unique features as auto tuning of params, being resistant to wind, etc) in 0.5.12 (check out livestream, 8 min mark) https://www.pscp.tv/w/1OyKApqkPyzxb?t=1s -- I'm closing this issue.

Fingers crossed we get some cool features to play with:
https://arxiv.org/pdf/1701.07254.pdf
https://fenix.tecnico.ulisboa.pt/downloadFile/563345090414617/RA_EduardoLSSilva_69916.pdf

@pd0wm Looking forward to your implementation!
[image]

👍

closed 2019-05-09 by zorrobyte

Jrpmarquet, 2023-03-17:
> Stepping away from stock code, let me know in Discord how you like my latest VSR port.

Hello Erich, on related subject I am looking to purchase the steering sensor kit that I was referred to you for? I have prius prime 2018 with ping pong issue. It is driving me crazy but I otherwise love the openpilot system. If you can help please email me at jrpmarquette@gmail.com