commaai/openpilot #602: Support 2019 Lexus RX450h with Color HUD?

schomems, 2019-04-14:
Can support be added for 2019 Lexus RX450h with Color HUD? So far OP will not identify car. Fingerprint info below:

fingerprint 36: 8, 37: 8, 166: 8, 170: 8, 180: 8, 295: 8, 296: 8, 426: 6, 452: 8, 466: 8, 467: 8, 550: 8, 552: 4, 560: 7, 562: 6, 581: 5, 608: 8, 610: 5, 643: 7, 658: 8, 713: 8, 740: 5, 742: 8, 743: 8, 800: 8, 810: 2, 812: 3, 814: 8, 818: 8, 819: 8, 820: 8, 821: 8, 822: 8, 830: 7, 835: 8, 836: 8, 845: 5, 863: 8, 865: 8, 869: 7, 870: 7, 871: 2, 898: 8, 900: 6, 902: 6, 905: 8, 913: 8, 918: 8, 921: 8, 933: 8, 944: 8, 945: 8, 950: 8, 951: 8, 953: 8, 955: 8, 956: 8, 971: 7, 975: 6, 993: 8, 998: 5, 999: 7, 1000: 8, 1001: 8, 1014: 8, 1017: 8, 1020: 8, 1041: 8, 1042: 8, 1044: 8, 1056: 8, 1057: 8, 1059: 1, 1063: 8, 1071: 8, 1076: 8, 1077: 8, 1082: 8, 1114: 8, 1161: 8, 1162: 8, 1163: 8, 1164: 8, 1165: 8, 1166: 8, 1167: 8, 1227: 8, 1228: 8, 1235: 8, 1237: 8, 1263: 8, 1264: 8, 1279: 8, 1349: 8, 1350: 8, 1351: 8, 1413: 8, 1414: 8, 1415: 8, 1416: 8, 1552: 8, 1553: 8, 1556: 8, 1557: 8, 1568: 8, 1570: 8, 1571: 8, 1572: 8, 1575: 8, 1592: 8, 1594: 8, 1595: 8, 1649: 8, 1777: 8, 1779: 8

rbiasini, 2019-04-14:
@schomems , definitely. Could you please post (or PM me on discord) a cabana drive with stock system operating?

schomems, 2019-04-14:
Hi rbiasini! I haven't done a drive yet since the EON / Panda won't find the car and the car is giving nothing but errors currently. How can I find you on discord? I searched for your handle here but found nothing there. I can PM you further details there. Thank you!

schomems, 2019-04-14:
Hi again rblasini! I just completed two drives- one with switches set to 0111 to enable OP, and another with switches set to 1111 to set to stock. I tried to replicate the drive as exactly as possible. Can you DM me on discord and I can give you any details you might need? Thank you!

ucolchen, 2019-04-14:
for toyota
1011 (Without comma power)
1001 (With comma power)
To use stock fuction

vanillagorillaa, 2019-04-24:
Merged in 5.11, closing

closed 2019-04-24 by vanillagorillaa