commaai/openpilot #603: Don't use OSM speed data due to accuracy issues - HD Maps enhancement

zorrobyte, 2019-04-14:
OSM, while an awesome project, frequently has incorrect speed data in my state of Indiana. This leads to unexpected behavior such as the vehicle suddenly attempting to slow to 35mph on a 70MPH interstate with no other roads around. Automated speed based features are unusable for me with OpenPilot due to this.

- OSM does not have a native speed layer, or a user-friendly speed contribution interface (personal opinion)
- OSM data can be maliciously modified

Instead, Comma should use aggregated GPS data to build a speed limit layer on HD Maps. OSM data can be initially imported to HD Maps, but then automatically enhanced by triggering a comma explorer annotation if the max speed variable is greatly different than vehicle speed. With enough drivers, we can accurately and automatically record speed limits without user intervention.

martinl, 2019-04-18:
or you could fix things when if they are broken and improve OSM data for everyone. OSM has a nice web-based editor with tutorial to get started with editing: https://www.openstreetmap.org

Setting a speed limit for a existing road is easy as clicking on the road and entering speed limit. Afaik, there are also OSM regional reviewers who will revert obviously incorrect changes.

Martin

zorrobyte, 2019-05-10:
Closing, as I've progressed in my OP knowledge and this issue no longer applies with the eventual release? of HD Maps

closed 2019-05-10 by zorrobyte